UVVM/UVVM #1: Configurable assertion time for AXI4Lite BREADY/RREADY signals

abyszuk, 2016-06-28:
I would like to be able to configure when BREADY and RREADY signals are asserted, in similar way that *VALID signals (e.g. config.num_b_pipe_stages).
I've noticed that there is a field in configuration record, but there's no support in procedures.

UVVM, 2016-06-29:
At the moment we have no resources to make these changes, as vacation is coming up and we need to prioritise other features on our UVVM roadmap. 
We do hope that the Open Source VHDL community will embrace UVVM and make new and improved VVCs/BFMs available, so maybe this is a great opportunity to get started. 
Otherwise I guess we could make the AXI4Lite improvements some time in Q3 or Q4, but again a bit depending on priorities.
We have put this into our task management system.

abyszuk, 2016-06-29:
OK, so if I were to prepare relevant patches you'd be willing to merge them? How should I proceed with this?
Typical pull request via a forked repo?

UVVM, 2016-06-29:
Yes – provided we do not think this is a major risk.
Your suggested approach sounds fine. Alternatively send your suggested version to support@bitvis.no<mailto:support@bitvis.no> and we will merge it before our next update.

-Espen

closed 2016-08-23 by UVVM