UVVM/UVVM #2: Wishbone BFM

abyszuk, 2016-06-28:
It would be really nice to have a Wishbone BFM supporting either classic or pipelined mode.
I think that "basic" set of signals (DAT_O, DAT_I, WE, SEL, CYC, STB, STALL, ACK, ERR) would be sufficient for most users.

UVVM, 2016-06-29:
At the moment we have no resources to prioritise this, but we have put this into our task management system for future evaluation. 
It would be great if the Open Source VHDL community would embrace UVVM and make this BFM/VVC.  
Otherwise maybe we could do this some time in Q3 or Q4, A Wishbone classic BFM and VVC is relatively simple to make - especially if starting from the SBI BFM/VVC.

UVVM, 2016-06-29:
(Test of direct answer on Github notification email)

At the moment we have no resources to prioritise this, but we have put this into our task management system for future evaluation.
It would be great if the Open Source VHDL community would embrace UVVM and make this BFM/VVC.

Otherwise maybe we could do this some time in Q3 or Q4, A Wishbone classic BFM and VVC is relatively simple to make - especially if starting from the SBI BFM/VVC.

closed 2016-08-23 by UVVM