UVVM/UVVM #3: Use multiple instances of a VVC concurrently

matteovit, 2016-08-16:
Hi,
I have a design with multiple instances of the same interface. 
The testbench has N instances of the same VVC (each one with a different GC_INSTANCE_IDX) and I can use each VVC from one single process, one VVC at time. 
But if I try to use all the VVCs at the same time (one process doing VVC calls for each VVC) I get an error about missing resolution function for the *_VVCT (t_vvc_target_record) signal.
I think the problem is that the *_VVCT signal is defined in a package (vvc_methods_pkg) and hence it is shared across all the VVCs.
Have you encountered the same problem before? 

Thanks,
Matteo

UVVM, 2016-08-16:
Hi Matteo,

The VVCs in UVVM are intended to be controlled from a single central sequencer (VHDL process). This in order to have a single brain in the system and to avoid the need for additional inter process (or inter entity/component) synchronization.
If I understand you correctly you have multiple processes controlling one or more VVCs. This will not work with UVVM as it is today – as this was originally something we wanted to avoid. Is there any way you can change your testbench to allow all VVC control from a single process?

- Espen

matteovit, 2016-08-16:
Hi Espen,
thank you for your reply. 
Let's assume my design has N UARTs and I have N UART VVCs. At the moment I have a single process that has this code:
```
for uart_index in 0 to N-1 loop
  uart_expect(UART_VVCT, uart_index+1, ..)
  await_completion(UART_VVCT, uart_index+1, ..)
end loop;
```
which is not ideal because it is testing one UART at time (let's assume the UARTs are continuously sending data). So it takes a long simulation time and it is not a realistic scenario.

My first solution was to have N processes with each one calling uart_expect(UART_VVCT, uart_index+1, ..) and await_completion(UART_VVCT, uart_index+1, ..) but then the signal UART_VVCT is shared across all the VVCs.

I guess another option is to keep a single process and call uart_expect for all the N UARTs first; then await for completion. Something like:
```
for uart_index in 0 to N-1 loop
  uart_expect(UART_VVCT, uart_index+1, ..)  
end loop;

for uart_index in 0 to N-1 loop
  await_completion(UART_VVCT, uart_index+1, ..)
end loop;
```
I guess if i.e. VVC 2 completes before VVC 1, when await_completion is called for VVC 2, it will exit immediately. Is this correct?

Thank you,
Matteo

UVVM, 2016-08-17:
Hi Matteo,

Yes, your latter single process option below will do the trick.
And yes, await_completion() will return immediately if the relevant VVC is already finished, which means it can be looped as you suggest.

This allows you to do parallel testing of your UARTs with a realistic scenario.
My personal opinion is that having these loops in one single test sequencer (process) improves the overview – as you only need to check one single process to see what your complete verification environment is set up to do. I do however understand that there might be good reasons for other architectures as well.

As I don’t see the remaining parameters of your uart_expect() I don’t know if you have done this already, but it is very easy to extend the uart_expect() method for the VVC to expect an array of bytes rather than a single byte. You could also make commands to check against a shared variable of byte array or the provided data buffers, or slightly more advanced – have the transmitter VVC generate and transmit random data, with random delay,  - put these into a buffer or shared variable and check against this with uart_expect(). One of my presentations at FPGA-Kongress in Munich in July (see attachment) shows a potential syntax for this on pages 30-32.
I will cover this again in my tutorial at DVCon-Europe October 19.
In our courses ‘Accelerating FPGA VHDL Verification’ (Germany and Sweden this autumn) we cover this in detail in presentations and labs (see www.bitvis.no<http://www.bitvis.no> > Events).

BTW - we will soon introduce multicast and broadcast for UVVM. This will allow your second loop (of your latter example) to be reduced to a single await_completion() with no need for a loop.

Regards
Espen

matteovit, 2016-08-18:
Hi Espen,
I have implemented the single process version with one loop for recv and another loop for await and it works perfectly. 

Regarding more complex data than a single byte, I have a custom protocol VVC where I can pass a record with a complete data packet to the send or check procedure. 
Your I2C VVC is also a good example with overloaded methods for single byte or multiple bytes transactions.

Unfortunately there isn't an attachment to your message. 

I had a look at your courses and they seem interesting, unfortunately I have already booked all my conferences/events for the rest of 2016!

Regards,
Matteo

UVVM, 2016-08-23:
Hi Matteo,
For some strange reason I never got an email notification on your last comment. 

Good to hear that the new solution works fine
- and that you found an example on making overloads for multiple byte transactions :-)

It seems sending attachments doesn't work when answering email via GitHub.
Send your email-address to support@bitvis.no and I will send the file directly to you.

Regards
Espen

closed 2016-08-23 by UVVM