UVVM/UVVM #7: Critical warnings from GHDL: Pure function "..." cannot call (impure) procedure "...".

Paebbels, 2016-11-11:
I got further in integrating UVVM into one of our testbenches.

**GHDL unveils more critical warnings:**
> uvvm_util\src\bfm_common_pkg.vhd:139:15:warning : pure function "normalize_and_check" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:142:15:warning : pure function "normalize_and_check" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:150:17:warning : pure function "normalize_and_check" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:153:17:warning : pure function "normalize_and_check" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:164:17:warning : pure function "normalize_and_check" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:202:15:warning : pure function "normalize_and_check" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:205:15:warning : pure function "normalize_and_check" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:213:17:warning : pure function "normalize_and_check" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:218:19:warning : pure function "normalize_and_check" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:223:21:warning : pure function "normalize_and_check" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:240:17:warning : pure function "normalize_and_check" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:265:14:warning : pure function "normalise" cannot call (impure) procedure "deprecate"
> uvvm_util\src\methods_pkg.vhd:279:13:warning : (procedure "deprecate" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:268:15:warning : pure function "normalise" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:271:15:warning : pure function "normalise" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:279:17:warning : pure function "normalise" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:282:17:warning : pure function "normalise" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:293:17:warning : pure function "normalise" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:329:14:warning : pure function "normalise" cannot call (impure) procedure "deprecate"
> uvvm_util\src\methods_pkg.vhd:279:13:warning : (procedure "deprecate" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:332:15:warning : pure function "normalise" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:335:15:warning : pure function "normalise" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:343:17:warning : pure function "normalise" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:348:19:warning : pure function "normalise" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:353:21:warning : pure function "normalise" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:370:17:warning : pure function "normalise" cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:121:12:warning : pure subprogram body cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:184:12:warning : pure subprogram body cannot call (impure) procedure "tb_error"
> uvvm_util\src\methods_pkg.vhd:205:13:warning : (procedure "tb_error" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:249:12:warning : pure subprogram body cannot call (impure) procedure "deprecate"
> uvvm_util\src\methods_pkg.vhd:279:13:warning : (procedure "deprecate" is defined here)
> uvvm_util\src\bfm_common_pkg.vhd:313:12:warning : pure subprogram body cannot call (impure) procedure "deprecate"
> uvvm_util\src\methods_pkg.vhd:279:13:warning : (procedure "deprecate" is defined here)

**Finally I get this error message:**
> uvvm_vvc_framework\src_target_dependent\td_queue_pkg.vhd:21:9: body of package "td_queue_pkg" was never analyzed

I'll investigate in the next days if this error is caused by GHDL itself or by UVVM.

andrefi, 2016-12-08:
Hello, and sorry for the late reply. 
You are correct about the impure functions. This will be fixed in the next release.

I'm not quite sure about the error you got. Did you manage to solve it? Are you trying to compile one of our VIPs?

Paebbels, 2016-12-08:
The error was a bug in GHDL and got filed. I is (or should be) fixed soon.

closed 2017-04-04 by UVVM