UVVM/UVVM #8: How to check a single I²C address call with I2C_VVC?

Paebbels, 2016-11-13:
I found 2 check procedures for a I²C slave: With one data byte and with 4 data bytes. How can I check a simple address issued by the master to test for a slave response?

It's valid to the protocol to just address a slave, which even isn't required to acknowledge. This is sometimes called a quick command. It can carry up to 1 bit of information: The read/write bit. In addition to this this command can be used to discover the bus.

andrefi, 2016-12-08:
Hello,

The I2C Quick Command (or write of 0 bytes) is currently not supported by our BFM, but since it is allowed by the protocol we will add this feature as soon as possible. It will most likely be part of our next I2C VIP release. 

Regarding your other questions:
> How do I distinguish read and write operations on the bus by enqueuing check commands?
I'm not sure what you mean. Are you thinking from a I2C slave point of view, i.e. using i2c_slave_check?

> How How do I await restart conditions after n transferred bytes?
If I understand you correctly, you are talking about sending a block write, then a repeated start condition after N bytes then send another block of bytes?

Paebbels, 2016-12-08:
Hello Andre,

> > How do I distinguish read and write operations on the bus by enqueuing check commands?

> I'm not sure what you mean. Are you thinking from a I2C slave point of view, i.e. using i2c_slave_check?

For example, when a slave receives an address, it get's activated because the slave address is encoded via VVC configuration. How can I test in which mode the slave was activated? It can be either read or write mode.

> > How How do I await restart conditions after n transferred bytes?

> If I understand you correctly, you are talking about sending a block write, then a repeated start condition after N bytes then send another block of bytes?

I²C and especially SMBus and PMBus have so called call or execution protocols. All these protocols build on top of I²C, which only provides:
- start condition
- send byte
- read byte
- restart condition
- stop condition

The mentioned 0-byte transfer is assembles as:
1. start condition
2. send byte (address inkl. read/write bit)
3. stop condition

More complex protocols with e.g. device select, register select and 2 byte write are assembled as follows:

1. start condition
2. send byte (device address inkl. write bit)
3. send byte (register address)
4. send byte (data 0)
5. send byte (data 1)
6. stop condition

A 2 byte read with device and register select can be assembled so:

1. start condition
2. send byte (device address inkl. write bit)
3. send byte (register address)
4. restart condition
5. send byte (device address inkl. read bit)
6. receive byte (data 0)
7. receive byte (data 1)
8. stop condition

This form can be extended up to 257 write bytes and 257 read bytes with a bus turn around in the middle. This is known as execute or process call. You can send up to 255 parameter bytes and receive 255 result bytes. To check this pattern, I think it's required to test:
- for the restart condition
- for the re-transmitted device address

[SMBus 3.0](http://smbus.org/specs/SMBus_3_0_20141220.pdf) -> page 43 process call

andrefi, 2016-12-10:
Thank you for the clarification, Patrick. 
I will discuss among my colleagues regarding how to distinguish between read and write operations and get back to you as soon as possible.

Regarding the restart condition:
There restart condition is supported in our I²C VIP, at least from the master side. It can be used by setting the "continue" parameter to true, e.g.:
1. i2c_master_transmit procedure with N data bytes and continue = true
2. i2c_master_receive procedure with N data bytes and continue = false

If both master and slave were VVCs this would look something like this in your testbench:
`i2c_slave_check(I2C_VVCT, 1, v_bytes_from_master, "Master to Slave check, N bytes");
i2c_master_transmit(I2C_VVCT, 0, C_I2C_SLAVE_ADDR, v_bytes_from_master, "Master to Slave transmit, N bytes", false);
i2c_slave_transmit(I2C_VVCT, 1, v_bytes_from_slave, "Slave to Master transmit, N bytes");
i2c_master_check(I2C_VVCT, 0, C_I2C_SLAVE_ADDR, v_bytes_from_slave, "Slave to Master check, N bytes",true);
-- Await completion of all commands
await_completion(I2C_VVCT, 0, v_some_timeout);
await_completion(I2C_VVCT, 1, v_some_timeout);
`

The input type t_byte_array is unconstrained, so there are no restrictions to the number of bytes you can send and receive in these procedures.

I'm guessing that your DUT is an I²C master and you wish to use the slave receive procedures in the VVC, is that right? You want the slave VVC to wait until the master has transmitted N bytes, then have the VVC transmit N bytes back to the DUT?

By the way, we will look into the SMBus specification. This might be a good extension for our I²C VIP in the future.

Paebbels, 2016-12-10:
Hello Andre,

the DUT is a I²C master. It has one command and two FIFO interfaces. We can wrap this I2C controller in:
- a WishBone enabled I²C controller by adding a WishBone to FIFO adapter in between both buses
- a PicoBlazeBus enabled I²C controller
- a SMBus controller (translating SMBus commands to I²C commands)
- a MDIO over I²C controller (e.g. needed for Ethernet PHYs housed in a SFP cage)
- any device specific controller
  - e.g. the Si570 Any Frequency Generator can be manipulated by a "Si570 to I²C" adapter, which translates input frequencies (a simple table) to I²C commands and FIFO streams for our master

That's why our controller is so flexible and has a streaming (FIFO) interface.

--------------------------
The UVVM enhanced code I²C code is online available in a test branch:
- Sources: https://github.com/VLSI-EDA/PoC/tree/paebbels/uvvm_i2c/src/io/iic?ts=2
- Textbench: https://github.com/VLSI-EDA/PoC/tree/paebbels/uvvm_i2c/tb/io/iic?ts=2

andrefi, 2016-12-19:
Hello Patrick,

Your DUT sounds interesting, I'll have a closer look at it. 

I've looked into the issue regarding distinguishing between I²C read and write operations in our VVC. 
Two points here:

1. Our VVC was missing some functionality related to the I²C Quick Command, both from the Master side and from the Slave side (e.g. i2c_slave_check for transmissions without data bytes, but where the R/W# bit can be either '1' or '0'). This will be added to our next release, probably mid January. 

2. The slave VVC will automatically check if it was accessed in the correct "mode" when you call either i2c_slave_transmit ('1') or i2c_slave_check ('0'). The only exception is for the Quick Command, where you can use i2c_slave_check without the databyte, but instead specify the expected R/W# bit. 
Note that you need to know beforehand what you are expecting as output from your DUT. The VVC only works as a checker, not as a generic I²C slave model.

Paebbels, 2016-12-19:
Thanks Andre. I'm looking forward for the new release :).
Have nice Christmas holidays.

UVVM, 2019-04-26:
This feature was released in v2.0.3

closed 2019-04-26 by UVVM