UVVM/UVVM #9: Incorrect behaviour of tvalid in AXI4-Stream BFM transmit procedure when using independent tready assertion

higgns, 2017-01-20:
The AXI4-Stream BFM transmit procedure does not deassert `tvalid` correctly when the slave `tready` assertion is independent of the master `tvalid` signaling (as for example indicated by the third timing diagram on [this page](https://www.beyond-circuits.com/wordpress/tutorial/tutorial18/part4/)). This is imo covered by the AMBA specification.

A quick fix that worked for me is using `rising_edge(clk)` instead of `config.clock_period` when waiting for `tready`.

```vhdl
...
if v_wait_for_next_transfer_cycle then
    -- wait for config.clock_period;
    wait until rising_edge(clk);
    while axistream_if.tready = '0' loop
        wait for config.clock_period;
...
```
Let me hear your thoughts :).
Ben

UVVM, 2017-01-20:
Hi,

Thanks. You are absolutely right.

To your information UVVM 2.0 will be released very soon with some really interesting new features.
This bug has been fixed in UVVM 2.0.
A beta-version will be available already next week. Please let us know if you would like to test it out – prior to Github publishing.

Best regards
Espen

higgns, 2017-01-25:
Hi,

good to know. I'd love to try out the new version.

Ben

UVVM, 2017-01-25:
Hi,

You can have the beta version tomorrow (as a zip-file) if you send an email to info@bitvis.no<mailto:info@bitvis.no>.

- Espen

UVVM, 2017-04-04:
This was fixed in UVVM 2.0 released in February and thus closed.

closed 2017-04-04 by UVVM