UVVM/UVVM #10: avalon_mm_bfm_pkg readdatavalid, wrong data

soundart, 2017-01-31:
Hi,
there seems to be a problem in the avalon_mm_bfm if you configure it with use_readdatavalid        => true.

If avalon_mm_if.readdatavalid is high the bfm waits another rising edge and a bit and the samples avalon_mm_if.readdata.

I fixed this by commenting out the waits....

@@ -581,8 +582,8 @@ package body avalon_mm_bfm_pkg is
       if timeout then
         alert(config.max_wait_cycles_severity, proc_call & "=> Failed. Timeout waiting for readdatavalid");
       else
-        wait until rising_edge(clk);
-        wait_until_given_time_after_rising_edge(clk, config.clock_period/4);
+        --wait until rising_edge(clk);
+        --wait_until_given_time_after_rising_edge(clk, config.clock_period/4);
       end if;
 
     else  -- not waitrequest or readdatavalid - issue read, wait num_wait_states before finishing the read

Could you check please?
kind regards
  Frank

UVVM, 2017-01-31:
Hi Frank,

You are absolutely correct.
This is changed in UVVM 2.0 to be released very soon.

Best regards
Espen

soundart, 2017-01-31:
Hi Epsen,

thank you for the confirmation.

kind regards
  Frank

soundart, 2017-02-04:
Workaround in git thanks!

closed 2017-02-04 by soundart