UVVM/UVVM #11: license situation

soundart, 2017-02-01:
Hi,

I have a question wrt the license situation. From the github project page "./README.md" and "./LICENSE" I had the impression that the whole package is covered by a MIT license.

I have found now the following files which indicate other licenses:

```
./uvvm_osvvm/LICENSE.TXT
./bitvis_uart/License.txt
./uvvm_util/LICENSE.txt
./uvvm_util/src/license_pkg.vhd
./LICENSE
./bitvis_vip_i2c/LICENSE.TXT
./bitvis_vip_axistream/License.txt
./bitvis_vip_sbi/License.txt
./bitvis_irqc/LICENSE.txt
./bitvis_vip_avalon_mm/LICENSE.TXT
./bitvis_vip_uart/License.txt
./uvvm_vvc_framework/LICENSE.txt
./bitvis_vip_axilite/LICENSE.TXT
```

So if I would now want to create a avalon_streaming component, based on the bitvis_vip_axistream component is it ok to do the following:

a) name it bitvis_vip_avalonstream
b) name it soundart_vip_avalonstream
c) publish my new component  on github under MIT license
d) sell it to the highest bidder

kind regards
  Frank

UVVM, 2017-02-03:
Hi Frank,
All of UVVM and Bitvis VIPs are currently covered by the MIT license. (OSVVM has a different license)
To your questions:
a) It wouldn't be a good idea to call it bitvis_<something> as that would indicate that it came from Bitvis.
b,c,d)  Any/all of these would be perfectly OK. In fact it would be great for the VHDL community if we could all make lots of different VVCs like that. So - yes, we strongly ecourage this.
We have made our current VVCs available for free, and it would be great to have many more free VVCs, but it is also perfectly OK to make commercially available VVCs.

Best regards
Espen

soundart, 2017-02-04:
Hi Espen,

thank you for the clarification. 

kind regards
  Frank

closed 2017-02-04 by soundart