UVVM/UVVM #12: What is the recommended value for max-delta-cycles to use UVVM?

Paebbels, 2017-02-18:
UVVM needs a lot of iterations in a time step. So what is the recommended value for max-delta-cycles?

UVVM, 2017-02-22:
The default is 1000. We recommend to set it to 5000.
(If you have an extreme system - pushing several hundred commands at the same time, you may need to go even higher. Wouldn't recommend such a test sequencer though ;-)

From Modelsim User Manual (assume similar setting is available in all simulators):
_The iteration limit default value is 1000. If you receive an iteration limit warning, first increase
the iteration limit and try to continue simulation. You can set the iteration limit from the
Simulate > Runtime Options menu or by modifying the IterationLimit variable in the
modelsim.ini. See modelsim.ini Variables for more information on modifying the modelsim.ini
file._

Paebbels, 2017-02-22:
Thanks.

----------
Btw. GHDL can already compiled a few simple UVVM examples. So the new GHDL feature (unconstrained records) is working. We are now adding more tests.

closed 2017-02-22 by Paebbels