UVVM/UVVM #15: Modelsim/Questa workaround in axistream_vvc.vhd

fjaelltrollet, 2017-03-24:
Hi

In axistream_vvc.vhd, the commit with tag v2.1.0 replaced the record-port version of the axistream_transmit-call with a discrete call version in the TRANSMIT case. The reason is that SIGSEGV mentioned in the comment above.

I guess the cases RECEIVE and EXPECT require the same fix. I currently comment those cases, because they trigger the same SIGSEGV. I tried this with Questa 10.3d and 10.5c

Thanks for the library!
Best regards,
emanuel

fjaelltrollet, 2017-03-24:
My mouse is faster than me...

UVVM, 2017-03-27:
Hi Emanual,

Thank you for the feedback. 

Strange, we didn't get the fatal error on the receive and expect methods. We'll look into why the record-port versions causes fatal errors in Questa as soon as possible. 
We have added a similar fix for receive and expect as we did for transmit, so this should work fine while we investigate further.

Best regards
Andre

fjaelltrollet, 2017-03-28:
Hi Andre

Thanks, your fix works for me.
Do you report such bugs to Mentor? Or should I file a bug?

Best regards,
emanuel

UVVM, 2017-03-28:
Hi Emanuel,

It would be great if you could report it.
Tool users tend to get more attention ;-)

Best regards
Espen

fjaelltrollet, 2017-04-04:
Ok, I finally filed a SR. Let's see what happens...

BR/emanuel

UVVM, 2019-04-26:
Hi Emanuel,

Was this issue resolved?

Best regards,
Erick

fjaelltrollet, 2019-04-29:
Hi Erick

I recently checked the SR - I did not get any update there, the SR is still open.
I have to admit that I'm not using UVVM since quite some time, so I haven't tested it either. I'll try to find some time for test with newer questa versions.

Best regards,
emanuel

UVVM, 2019-04-29:
Hi Emanuel,

It would be great if you could let us know if it worked whenever you get time to test it.

Best regards,
Erick

UVVM, 2019-08-12:
This issue has been fixed in Questa. Tested with version 2019.1
We will change to the original record-version of the procedure calls in the next release.

closed 2019-08-12 by UVVM