UVVM/UVVM #16: IRQC Testbench Simulation fails with GHDL due to XX's

grilialex, 2017-04-01:
The last days I was trying to compile and test run UVVM with GHDL V0.34dev. I started with the IRQC model and after figuring out the correct compilation sequence simulated the example. Unfortunately the simulation ended prematurely due to XX's coming out from dout signal.

**Setup:**

- I got the latest GHDL (c94da9d37438) and 
- UVVM (103a0abc4b4b) (with git extension on Hg-mercurial).
- Compiled GHDL with MSYS2 and MinGW32-w64.
- Then using an MSYS2 window, I got to the folder of UVVM\bitvis_irqc\sim\.
- Then run [comp.bat](https://github.com/UVVM/UVVM_All/files/887876/test_comp.bat.txt) (added .txt extension to attach here).

**Problem:**
Simulation stops with an error. The [sim_fail.log.txt](https://github.com/UVVM/UVVM_All/files/887875/sim_fail.log.txt) shows the error message. See the [waveform](https://cloud.githubusercontent.com/assets/736016/24581380/cde236ec-1722-11e7-931e-88d8f356c5ea.jpg) below.

[image]

**Resolution**
It seems that changing the following line in the port definition of irqc_pif.vhd:
    dout : out std_logic_vector(7 downto 0)   := (others => '0');
to:
    dout : out std_logic_vector(7 downto 0)   := (others => 'L');

resolves the problem as seen from the [log](https://github.com/UVVM/UVVM_All/files/887874/sim_ok.log.txt) file and the [waveform](https://cloud.githubusercontent.com/assets/736016/24581415/783b223e-1723-11e7-8d7d-97765b9f15c3.jpg)
[image]

Full simulation shown here:
[image]

**Alternate Test**
Trying to figure out if it was a glitch of GHDL I created a separate testbench, where the original code irqc_pif.vhd worked fine.

[image]

Here is the zip file with the simplified testbench.
[TestGHDL.zip](https://github.com/UVVM/UVVM_All/files/887883/TestGHDL.zip)

Seems to me that there should be an issue with UVVM (maybe in combination with GHDL?).

[sim_ok.log.txt](https://github.com/UVVM/UVVM_All/files/887874/sim_ok.log.txt)
[sim_fail.log.txt](https://github.com/UVVM/UVVM_All/files/887875/sim_fail.log.txt)
[test_comp.bat.txt](https://github.com/UVVM/UVVM_All/files/887876/test_comp.bat.txt)

Paebbels, 2017-04-01:
You don't need to analyse all UVVM files per simulation run. GHDL is shipped with precompile scripts, those include UVVM (2.0.x). You can reference the generated UVVM folder with option [-P](http://ghdl.readthedocs.io/en/new-documentation-structure/using/InvokingGHDL.html#cmdoption-ghdl-p).

I further don't recommend old batch files. These are deprecated since years.

You don't need to start GHDL from within a MinGW Bash. At first your are executing a foreign script file in Bash. At second you can run GHDL directly from Windows PowerShell by setting up a correct MinGW environment.

Regarding the shown assertion error, this is caused by UVVM by executing a failing assert statement.

grilialex, 2017-04-02:
Thank you for your response. I am aware of the scripts but initially when I used v0.33 the compilation of UVVM failed, so In order to find out out the problem I tried to compile manually. Finally I figured out that I need to compile myself V0.34dev. Thank you for the -P option. 

I have done many experiments/work in order to make GHDL work (I need to go back and review all my steps), and it seems that only MSYS GHDL knows the standard libraries path (probably a bad configuration option during compilation). Moving the analyzed libraries files under GHDL/lib folder does not seem to work.

I created these scripts to avoid manual entry and also for someone to easily replicate the issue with a minimum of configuration. Eventually I will create my own make or ANT files. The script files are dump (no parameters passing or any smart things inside). Is there a case that this procedure will affect simulation? 

Yes UVVM fails due to an assertion, however as I presume this code is tested and I am not sure if this happens for real (UVVM bug) or is a consequence of a missing step from my side.

Paebbels, 2017-04-02:
The newest (pre-)release of GHDl is 0.34-dev from [01.03.2017](https://github.com/tgingold/ghdl/releases/tag/2017-03-01). It was build for several platforms with different backends. It also includes a new build for Windows (GNAT GPL; no MSYS2 needed). A pre-release for MinGW64 is under construction.

The last question needs to be answered by UVVM. I'm just a UVVM tester and GHDL developer :).

UVVM, 2017-04-03:
Hi,
We will check the UVVM question.

Regards
Espen

UVVM, 2017-04-03:
Hi,
We had a look at this issue, and we suspect there is something wrong in GHDL.
There are two drivers for this signal: ‘dout <= dout_i’ inside irqc_pif and the BFM procedure inside p_main in the testbench.
p_main applies a ‘Z’, and hence cannot contribute to ‘X’ on an SLV (Standard_logic_vector)
dout is only driven by dout_i from process p_read_reg inside irqc_pif, but is initialised to zeros in the port declaration.
From the wave-view we can see that dout_i is correct, so it seems that GHDL is creating a third driver from the port declaration signal initialisation itself.
This also matches the fact that modifying the initialisation changes the behaviour.
The initialisation should not create an additional driver – as that would basically disallow initialisation…

Patrick_Could you check this in GHDL?

Regards
Espen

From: Support
Sent: Monday, April 3, 2017 2:24 PM
To: 'UVVM/UVVM_All'; UVVM/UVVM_All
Cc: Subscribed
Subject: RE: [UVVM/UVVM_All] IRQC Testbench Simulation fails with GHDL due to XX's (#16)

Hi,
We will check the UVVM question.

Regards
Espen

Paebbels, 2017-04-03:
Hello @grilialex.
Please open an issue at GHDL and link to this conversation.

Thanks.

grilialex, 2017-04-03:
Ok, done.
See [here](https://github.com/tgingold/ghdl/issues/328)

Confirming Espen, that observed that dout was not following dout_i (I tried to modify the code initially to figure out why this happens).

closed 2017-04-04 by UVVM