UVVM/UVVM #17: Questa strange vector width change

fjaelltrollet, 2017-04-04:
Hi

My TH has two axistream VVCs, one being 32 bits and one 64 bits. For one test, the TB first sends a frame to the 64 bit VVC, then a frame to the 32 bit VVC. 
When I just add another test after this, the 64 bit command fails with questa complaining about bitwidth issues in axistream_transmit (axistream_bfm_pkg): It sees some lhs as 32 bit and rhs as 64 bits.

Keeping the transmits in bitwidth-order (i.e. breaking the test and first sending two frames to 64 bits, then two frames to 32 bits), everything works fine.

Replacing the axistream_if record in axistream_transmit from axistream_bfm_pkg (the core function that is used by all others) with individual signals, the error is gone.

I assume
- that questa somehow fixes the vector length in the vector internally (-> questa bug)
- that you cannot do much about it
- that signal records and function calls in VHDL are poorly supported by Questa (see other issues)

I hope that I find some time to file another SR at mentor...

Best regards
emanuel

UVVM, 2017-04-04:
Hi Emanuel,

This looks very much like the other Questa bug, so I guess your assumption is correct.
Great if you file another SR at Mentor :-)

Best regards
Espen

UVVM, 2019-04-26:
Hi Emanuel,

Was this issue resolved?

Best regards,
Erick

fjaelltrollet, 2019-04-29:
Hi Erick

I did not file an SR here - expecting the other one to be handled first. But this didn't happen (see there).
Again, I'll try to test it with newer versions.

Best regards,
emanuel

UVVM, 2019-04-29:
Hi Emanuel,

That's alright, please keep us updated if you manage to test it.

Best regards,
Erick

UVVM, 2023-03-27:
Assume tool problem, but we will make a dedicated testbench to make sure.

closed 2024-05-02 by UVVM