UVVM/UVVM #18: VIP BFM AXI4-Lite with ALDEC Active-HDL 9.2 simulator

andrews6, 2017-05-09:
Hi,
I'm trying to use the VIP BFM AXI4-Lite with a proprietary testbench and methodology for a custom application. I'm running the testbench in ALDEC Active-HDL 9.2 simulator.
When I try to compile axilite_bfm_pkg.vhd I get this error from Active-HDL:

```
# Compile Package "axilite_bfm_pkg"
# Compile Package Body "axilite_bfm_pkg"
# Error: DAGGEN_0007: axilite_bfm_pkg.vhd : (0, 0): Error during conversion C:\Users\ADMINI~1\AppData\Local\Temp\7512_349979928axilite_bfm_pkg_axilite_bfm_pkg.dag to C:\Users\ADMINI~1\AppData\Local\Temp\7512_349979928axilite_bfm_pkg_axilite_bfm_pkg._x86.bin
# Compile failure 1 Errors 0 Warnings  Analysis time :  78.0 [ms]
```

And when I try to simulate I get:

```
# ELAB2: Elaboration final pass...
# ELAB2: Fatal Error: ELAB2_0030 Error while reading binary code for unit "axilite_bfm_pkg".
# KERNEL: Error: E8005 : Kernel process initialization failed.
# Error: Fatal error occurred during simulation initialization.
```

I am sure there are no syntax error in the file. I am compiling with VHDL-2008. Can you help me? Is Active-HDL supported? If not, why it shouldn't? 

Thanks, Andrea
[image]
[image]

UVVM, 2017-05-09:
Hi Andrea,

The code is checked with Modelsim (Mentor) and Riviera Pro (Aldec), but not with Active HDL.
I have informed Aldec about this issue.

Best regards
Espen

UVVM, 2017-05-17:
Hi Andrea,

According to Aldec your problem was caused by an old version of Active HDL, so this should not be a problem with the newest version.
We also fixed a problem in UVVM on Tuesday that might have led to another (and relevant )compile error.

Hopefully everything will work now with an updated version of the compiler.

Best regards
Espen

closed 2017-10-17 by UVVM