UVVM/UVVM #19: UVVM 2.3.0 contains invalid VHDL code

Paebbels, 2017-05-11:
Hello,

@andresdemski reported an issue ([#343 - Compiling UVVM with compile-uvvm.sh failed](https://github.com/tgingold/ghdl/issues/343)) in GHDL's [pre-compile scripts for UVVM](https://github.com/tgingold/ghdl/blob/master/libraries/vendors/compile-uvvm.sh). A part of that issue was discussed in a separate issue report ([#345 - Reading out parameters in procedures is not supported](https://github.com/tgingold/ghdl/issues/345)), but by reading the VHDL-2008 LRM again, we figured out, that the problem is not on the GHDL side.

Packages like `methods_pkg.vhd` contain invalid VHDL code e.g. at [line 5151](https://github.com/UVVM/UVVM_All/blob/v2.3.0/uvvm_util/src/methods_pkg.vhd?ts=2#L5151). According to the compilation log file provided by @andresdemski, this issue occurs several times in that file and possibly also in other files of UVVM.

Subprogram parameters of kind **signal** and mode **out** can not be read.

Kind regards
    Patrick

-------------------------
Here is a quote from VHDL-2008 LRM proving that reading signal parameters is not valid:
> **LRM 6.5.2 Interface object declarations**
> ...
> An interface object has one of the following modes:
> * **in** ...
> * **out**. The value of the interface object is allowed to be updated and,  provided it is not a signal parameter, read. Reading the attributes of the interface object is allowed, unless the interface object is a signal parameter and the attribute is one of 'STABLE, 'QUIET, 'DELAYED, 'TRANSACTION, 'EVENT, 'ACTIVE, 'LAST_EVENT, 'LAST_ACTIVE, or 'LAST_VALUE.
> * **inout** or **buffer** ...
> * **linkage** ...

UVVM, 2017-05-17:
Hi Patrick,

Thanks for your feedback.
You are absolutely right.
The relevant clock generator procedure outputs have now been changed to inout.

Best regards
Espen

UVVM, 2017-10-17:
This issue has been fixed in the latest release.

closed 2017-10-17 by UVVM