UVVM/UVVM #20: Aldec Active-HDL $version variable

cgaudin, 2017-05-18:
Hello,

The variable $version is pre-defined and used for the version number.
All the build scripts (compile_src.do, ...) need to be updated in order to avoid trouble.

Just adding an underscore avoid the variable collision in Active-HDL.

<--->
# Detect simulator
if {[catch {eval "vsim -version"} message] == 0} { 
  quietly set _version [eval "vsim -version"]
  # puts "Version is: $_version"
  if {[regexp -nocase {modelsim} $_version]} {
    quietly set simulator "modelsim"
  } elseif {[regexp -nocase {aldec} $_version]} {
    quietly set simulator "rivierapro"
  } else {
    puts "Unknown simulator. Attempting to use Modelsim commands."
    quietly set simulator "modelsim"
  }  
} else { 
    puts "vsim -version failed with the following message:\n $message"
    abort all
}
<--->

Regards,
Cédric

UVVM, 2017-05-19:
Hello Cédric,

- and thanks for your feedback.
We will change this to allow simpler use with Active HDL.

Best regards
Espen

UVVM, 2017-06-08:
Hi Cédric,
We have updated the compile scripts with new variable name. 
Again, thank you for your feedback.

Br,
Marius

closed 2017-06-08 by UVVM