UVVM/UVVM #21: Vendor and tool independent compile order files

Paebbels, 2017-05-22:
Hello,

it would be nice if UVVM would provide vendor and tool independent compile order files. Compile order files are simple text files containing one VHDL filename per line (ideally as a relative path to the root directory).

This would allow for instance PoC or GHDL to adapt it's UVVM pre-compile scripts faster to UVVM changes. All VIP pre-compilations would be independent of UVVM changes. Only new VIPs need to be added to a list of VIPs in such a pre-compile script.

Such compile order files could also be used in UVVMs TCL files and replace all calls to `vcom` by a generic solution. Changing files sets of a VIP would then be limited to changing the compile order file in stead of adjusting the TCL files.

Other users of such compile order files: Xilinx ISE, Xilinx Vivado

UVVM, 2017-05-29:
Hi Patrick,

Thanks for your input.
We will have a closer look at this.

Best regards
Espen

UVVM, 2017-06-08:
Hi Patrick.
We have added the compile order files to the script folder.

Br,
Marius

closed 2017-06-08 by UVVM