UVVM/UVVM #22: VVC Name to long

asicnet, 2017-06-13:
The max length of VVC name is 14. vcc_generator should display the max value.
C_LOG_SCOPE_WIDTH is 20 : 4 character for "_VVC" and 2 for ',' and number.
vr_scope : string (1 to 30) is to small if C_LOG_SCOPE_WIDTH is only changed , as the generator suggests
Suggestion: a new constant for the VVC_NAME length

adaptation_pkg.vhd
  constant C_VVC_NAME_WIDTH    : natural := 14;
  constant C_LOG_SCOPE_WIDTH   : natural := C_VVC_NAME_WIDTH+4+2; -- "_VVC" & ',' and number 

ti_generic_queue_pkg.vhd
variable vr_scope  : string(1 to C_LOG_SCOPE_WIDTH+13)   := (others => NUL);

UVVM, 2017-06-20:
Hi,

Thank you for reporting this issue.
Your suggestion will be implemented as soon as possible.

asicnet, 2017-06-27:
Hi
a correction 
The C_MAX_VVC_INSTANCE_NUM is 64 and so a 2char number

 constant C_LOG_SCOPE_WIDTH   : natural := C_VVC_NAME_WIDTH+4+1+2; -- "_VVC" & ',' and "2char number"

UVVM, 2017-08-08:
This issue have been addressed in a new significant update scheduled for Q3.
We have now introduced two new parameters: C_MINIMUM_VVC_NAME_SCOPE_WIDTH and C_MINIMUM_CHANNEL_SCOPE_WIDTH.
VVC name and channel name are linearly truncated based on length if not minimum length is reached to satisfy C_LOG_SCOPE_WIDTH when printed in transcript.
Max length for VVC name in vvc_generator is removed.

UVVM, 2017-10-17:
This issue has been fixed in the latest release.

closed 2017-10-17 by UVVM