UVVM/UVVM #23: Issue with i2c_slave_receive procedure when I2C VIP used as Master and Slave in a Test Harness

AHHarish, 2017-07-11:
Following instances are used in a Test Harness, no additional instantiations done. This plaform is created to check Master and Slave behaviors of the I2C VIP.
  i1_i2c_vvc: entity bitvis_vip_i2c.i2c_vvc
  generic map(
    GC_MASTER_MODE  => true,
    GC_INSTANCE_IDX   => 1,
    GC_I2C_CONFIG                            => C_I2C_BFM_CONFIG_DEFAULT,  -- Behavior specification for BFM
    GC_CMD_QUEUE_COUNT_MAX                   => 1000,
    GC_CMD_QUEUE_COUNT_THRESHOLD             => 950,
    GC_CMD_QUEUE_COUNT_THRESHOLD_SEVERITY    => warning,
    GC_RESULT_QUEUE_COUNT_MAX                => 1000,
    GC_RESULT_QUEUE_COUNT_THRESHOLD          => 950,
    GC_RESULT_QUEUE_COUNT_THRESHOLD_SEVERITY => warning
  )
  port map(
    i2c_vvc_if        =>  i2c_vvc_if
  );

  i2_i2c_vvc: entity bitvis_vip_i2c.i2c_vvc
  generic map(
    GC_MASTER_MODE  => false,
    GC_INSTANCE_IDX   => 2,
	GC_I2C_CONFIG                            => C_I2C_BFM_CONFIG_DEFAULT,  -- Behavior specification for BFM
    GC_CMD_QUEUE_COUNT_MAX                   => 1000,
    GC_CMD_QUEUE_COUNT_THRESHOLD             => 950,
    GC_CMD_QUEUE_COUNT_THRESHOLD_SEVERITY    => warning,
    GC_RESULT_QUEUE_COUNT_MAX                => 1000,
    GC_RESULT_QUEUE_COUNT_THRESHOLD          => 950,
    GC_RESULT_QUEUE_COUNT_THRESHOLD_SEVERITY => warning
  )
  port map(
    i2c_vvc_if        =>  i2c_vvc_if
  );

In a testbench used following two commands
i2c_slave_receive(I2C_VVCT, 2, 1, "One byte from master to slave");

i2c_master_transmit(I2C_VVCT,1,C_FPGA_DEVICE_ID,  x"AA", "I2C TX", FALSE);

Following is the error GOT when the Master is about to generate stop condition. When debugged @ low level, found i2c_slave_receive procedure checking the timeout 20ns check before Master is able to assert STOP condition.
UVVM: ***  FAILURE #1  ***
UVVM:        178240 ns   I2C_VVC,2
UVVM:                    await_value(std_logic 1, 0 ns, 20 ns) => Failed. Timed out after 20 ns. 'One byte from master to slave'  [1]

Let me know if there is any mistake in the usage.

UVVM, 2017-07-20:
Hi, 
Thank you for using UVVM!
Is it possible for you to supply your code? And what version of UVVM are you using?

Br,
Marius

AHHarish, 2017-08-07:
Following is the Version Details.
This project uses the following versions:
  uvvm_vvc_framework, version v2.0.4
  uvvm_util, version v2.0.4
  bitvis_vip_sbi, version v2.0.4
  bitvis_vip_uart, version v2.0.3
  bitvis_uart, version v1.0.3
  bitvis_irqc, version v1.0.3
  bitvis_vip_axilite, version v2.0.5
  bitvis_vip_axistream, version v3.0.2
  bitvis_vip_avalon_mm, version v2.0.5
  bitvis_vip_i2c, version v2.0.3
  bitvis_vip_spi, version v2.0.1

The TB and TH code is already there in the previous.

library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

library uvvm_util;
context	uvvm_util.uvvm_util_context;
library uvvm_vvc_framework;
use uvvm_vvc_framework.ti_vvc_framework_support_pkg.all;
--use work.i2c_bfm_pkg.all;

use work.timing_const_pkg.all;

library bitvis_vip_i2c;
library bitvis_i2c;
use	bitvis_vip_i2c.i2c_bfm_pkg.all;
use	bitvis_vip_i2c.vvc_methods_pkg.all;
-- Test harness entity
entity i2c_vvc_th is
end entity;

-- Test harness architecture
architecture struct of i2c_vvc_th is

  -- I2C VVC signals
  signal  i2c_vvc_if    : bitvis_vip_i2c.i2c_bfm_pkg.t_i2c_if;
  --signal  i2c_vvc_if    : t_i2c_if;
  
  
begin

  -----------------------------------------------------------------------------
  -- Instantiate the concurrent procedure that initializes UVVM
  -----------------------------------------------------------------------------
  i_ti_uvvm_engine : entity uvvm_vvc_framework.ti_uvvm_engine;

  -----------------------------------------------------------------------------
  -- I2C VVC
  -----------------------------------------------------------------------------  
  i1_i2c_vvc: entity bitvis_vip_i2c.i2c_vvc
  generic map(
    GC_MASTER_MODE  => true,
    GC_INSTANCE_IDX   => 1,
    GC_I2C_CONFIG                            => C_I2C_BFM_CONFIG_DEFAULT,  -- Behavior specification for BFM
    GC_CMD_QUEUE_COUNT_MAX                   => 1000,
    GC_CMD_QUEUE_COUNT_THRESHOLD             => 950,
    GC_CMD_QUEUE_COUNT_THRESHOLD_SEVERITY    => warning,
    GC_RESULT_QUEUE_COUNT_MAX                => 1000,
    GC_RESULT_QUEUE_COUNT_THRESHOLD          => 950,
    GC_RESULT_QUEUE_COUNT_THRESHOLD_SEVERITY => warning
  )
  port map(
    i2c_vvc_if        =>  i2c_vvc_if
  );

[... 114 lines cut ...]
    -- the transmission, the SBI VVC is instructed to check read and check (sbi_check) the C_ADDR_RX_DATA 
    -- register, and verify that it is in fact x"AA" that the DUT received. The test sequencer will continue
    -- when the SBI VVC is done checking the C_ADDR_RX_DATA register.

    i2c_slave_receive(I2C_VVCT, 2, 1, "One byte from master to slave");

    i2c_master_transmit(I2C_VVCT,1,C_FPGA_DEVICE_ID, x"AA", "I2C TX", FALSE);

   wait for 50 ns;  -- margin     

    -----------------------------------------------------------------------------
    -- Ending the simulation
    -----------------------------------------------------------------------------
   wait for 4000 ns;             -- to allow some time for completion
    report_alert_counters(FINAL); -- Report final counters and print conclusion for simulation (Success/Fail)
    log(ID_LOG_HDR, "SIMULATION COMPLETED", C_SCOPE);
    
    -- Finish the simulation

   std.env.stop;
    --wait;  -- to stop completely

  end process p_main;

end func;

UVVM, 2017-08-08:
Hi @AHHarish 
I've tested your design using bitvis_vip_i2c v2.0.5, uvvm_util v2.1.1 and uvvm_vvc_framework v2.0.7 releases, and the TB runs fine:

i2c_vvc_tb:
`library IEEE;
use IEEE.std_logic_1164.all;
use IEEE.numeric_std.all;

library uvvm_util;
context uvvm_util.uvvm_util_context;

library uvvm_vvc_framework;
use uvvm_vvc_framework.ti_vvc_framework_support_pkg.all;

library bitvis_vip_i2c;
use bitvis_vip_i2c.vvc_methods_pkg.all;
use bitvis_vip_i2c.vvc_cmd_pkg.all;
use bitvis_vip_i2c.td_vvc_framework_common_methods_pkg.all;
use bitvis_vip_i2c.td_target_support_pkg.all;
use bitvis_vip_i2c.i2c_bfm_pkg.all;

--use work.timing_const_pkg.all;

-- Test bench entity
entity i2c_vvc_tb is
end entity;

-- Test bench architecture
architecture func of i2c_vvc_tb is
  --constant RELEASE_LINE_AFTER_TRANSFER : boolean              := false;
  constant C_SLAVE_ADDR                : unsigned(6 downto 0) := "1010000";
  constant C_SCOPE                     : string               := C_TB_SCOPE_DEFAULT;

-- Clock and bit period settings
  constant C_CLK_PERIOD   : time := 60.098 ns;  -- 16.368 MHz
  constant C_CLOCK_PERIOD : time := 1000 ns;    --1MHZ

  constant C_BIT_PERIOD     : time                 := 16 * C_CLK_PERIOD;
  constant C_FPGA_DEVICE_ID : unsigned(9 downto 0) := "0001010000";
-- Log overload procedure for simplification
  procedure log(
    msg : string) is
  begin
    log(ID_SEQUENCER, msg, C_SCOPE);
  end;

begin
-- Instantiate test harness, containing DUT and Executors

  i_test_harness : entity work.i2c_vvc_th;
-- PROCESS: p_main

  p_main : process
    variable v_cmd_idx    : natural;    -- Command index for the last read
    variable v_byte_array : bitvis_vip_i2c.vvc_cmd_pkg.t_vvc_result;
  begin

-- Wait for UVVM to finish initialization
    await_uvvm_initialization(VOID);

-- Print the configuration to the log
    report_global_ctrl(VOID);
    report_msg_id_panel(VOID);

    enable_log_msg(ALL_MESSAGES);
--disable_log_msg(ALL_MESSAGES);
    enable_log_msg(ID_LOG_HDR);
    enable_log_msg(ID_SEQUENCER);
    enable_log_msg(ID_UVVM_SEND_CMD);

--disable_log_msg(I2C_VVCT, 1, ALL_MESSAGES);
--enable_log_msg(I2C_VVCT, 1, ID_BFM);

    log(ID_LOG_HDR, "Starting simulation of TB for I2C using VVCs", C_SCOPE);
------------------------------------------------------------

    log("Wait 10 clock period for reset to be turned off");
    wait for (10 * C_CLK_PERIOD);       -- for reset to be turned off

    log(ID_LOG_HDR, "Configure I2C VVC 1", C_SCOPE);
------------------------------------------------------------

-- shared_i2c_vvc_config(1).inter_bfm_delay.delay_in_time := 10 ms; --check if to add;
    shared_i2c_vvc_config(1).bfm_config.i2c_bit_time              := 10*C_BIT_PERIOD;
    shared_i2c_vvc_config(1).bfm_config.slave_mode_address        := C_FPGA_DEVICE_ID;
    shared_i2c_vvc_config(1).bfm_config.enable_10_bits_addressing := false;
--shared_i2c_vvc_config(2).inter_bfm_delay.delay_in_time := 10 ms; --check if to add;

    shared_i2c_vvc_config(2).bfm_config.i2c_bit_time              := 10*C_BIT_PERIOD;
    shared_i2c_vvc_config(2).bfm_config.slave_mode_address        := C_FPGA_DEVICE_ID;
    shared_i2c_vvc_config(2).bfm_config.enable_10_bits_addressing := false;
    log(ID_LOG_HDR, "Check simple receive", C_SCOPE);
------------------------------------------------------------
-- In this test case the UART VVC (TX channel) is instructed to send the data x"AA" to the DUT.
-- This data should be received and stored to a RX buffer by the DUT. After the UART VVC has completed
-- the transmission, the SBI VVC is instructed to check read and check (sbi_check) the C_ADDR_RX_DATA 
-- register, and verify that it is in fact x"AA" that the DUT received. The test sequencer will continue
-- when the SBI VVC is done checking the C_ADDR_RX_DATA register.

    i2c_slave_receive(I2C_VVCT, 2, 1, "One byte from master to slave");

    --i2c_master_transmit(I2C_VVCT, 1, C_FPGA_DEVICE_ID, x"AA", "I2C TX", false);
    i2c_master_transmit(I2C_VVCT, 1, C_FPGA_DEVICE_ID, x"AA", "I2C TX");

    wait for 50 ns;                     -- margin

-----------------------------------------------------------------------------
-- Ending the simulation
-----------------------------------------------------------------------------

    wait for 4000 ns;                   -- to allow some time for completion
    report_alert_counters(FINAL);  -- Report final counters and print conclusion for simulation (Success/Fail)
    log(ID_LOG_HDR, "SIMULATION COMPLETED", C_SCOPE);

-- Finish the simulation

    std.env.stop;
--wait; -- to stop completely

  end process p_main;

end func;
`

UVVM, 2017-08-08:
Hi @AHHarish 
I've tested your TB with our latest release. The simulation run fine with minor modifications:

**i2c_vvc_tb.vhd**
library bitvis_vip_i2c;
use bitvis_vip_i2c.vvc_methods_pkg.all;
use bitvis_vip_i2c.vvc_cmd_pkg.all;  -- ADDED
use bitvis_vip_i2c.td_vvc_framework_common_methods_pkg.all;
use bitvis_vip_i2c.td_target_support_pkg.all; -- ADDED
use bitvis_vip_i2c.i2c_bfm_pkg.all;
.......
  --constant RELEASE_LINE_AFTER_TRANSFER : boolean              := false;  -- REMOVED
.......
    i2c_master_transmit(I2C_VVCT, 1, C_FPGA_DEVICE_ID, x"AA", "I2C TX"); -- CHANGED

----------------------------------------------------------------------------------------
**i2c_vvc_th.vhd**
library bitvis_vip_i2c;
--library bitvis_i2c;  -- REMOVED
use bitvis_vip_i2c.i2c_bfm_pkg.all;
use bitvis_vip_i2c.vvc_methods_pkg.all;

-------------------------------------------------------------------------------------

Best regards,
Marius

closed 2017-08-08 by UVVM

AHHarish, 2017-08-09:
Hi Marius,

Please consider adding additional command as give below after the i2c_master_transmit command. Still it fails. Use the same setup what you have described. 

   i2c_slave_receive(I2C_VVCT, 2, 1, "One byte from master to slave");
   i2c_master_transmit(I2C_VVCT, 1, C_FPGA_DEVICE_ID, x"AA", "I2C TX"); -- CHANGED by Marius
   await_completion(I2C_VVCT,1,100 * 10*C_BIT_PERIOD); -- ADDED by AHHarish

Regards,
AHHarish