UVVM/UVVM #24: axistream_transmit procedure sending data even when vready is un-initialised

ansariabd, 2017-07-17:
Hi,

I have been using axistream and axilite BFMs for a while in my self checking simulations and have found out that the axistream_transmit procedure keeps sending the data to slave when vready signal un-initialised is. In general the ready signal from slaves must be in defined state, but since the purpose is verification of designs, I would like the developers to have a look into this procedure

on line : 367 in axistream_bfm_pkg.vhd

(while v_tready = '0' loop) could be changed with (while v_tready /= '1' loop).

UVVM, 2017-07-20:
Hi,
Thank you for the feedback!
We will look into this matter soon.

Best regards,
Marius

qarlosalberto, 2017-08-01:
@ansariabd have you an axilite bfm example? Im having problems with them :S

UVVM, 2017-08-04:
Hi Carlos.
In the src folder of bitvis_vip_axilite there is an axilite BFM already.

Best regards,
Marius

qarlosalberto, 2017-08-04:
@UVVM thank you.

ansariabd, 2017-08-04:
Hi Carlos,

If you just want axilite bfm package itself, then you can find it as
answered by UVVM.

But if you need the vhdl code where the procedures from axilite bfm package
are executed, then here is a small example.

library bitvis_vip_axilite;
use bitvis_vip_axilite.axilite_bfm_pkg.all;

after entity and achitecture

constant C_AXILITE_BFM_CONFIG_LOCAL : t_axilite_bfm_config := (
    max_wait_cycles            => 10,
    max_wait_cycles_severity   => TB_FAILURE,
    clock_period               => C_AXI_CLK_PERIOD,
    expected_response          => OKAY,
    expected_response_severity => TB_FAILURE,
    protection_setting         => UNPRIVILIGED_UNSECURE_DATA,
    num_aw_pipe_stages         => 1,
    num_w_pipe_stages          => 1,
    num_ar_pipe_stages         => 1,
    num_r_pipe_stages          => 1,
    num_b_pipe_stages          => 1,
    id_for_bfm                 => ID_BFM,
    id_for_bfm_wait            => ID_BFM_WAIT,
    id_for_bfm_poll            => ID_BFM_POLL
    );

  -- Signal for AXI4Lite Master Interface
  signal m_axilite_if : t_axilite_if (write_address_channel(awaddr(19
downto 0)), write_data_channel(wdata(31 downto 0), wstrb(3 downto 0)),
                        read_address_channel(araddr(19 downto 0)),
                        read_data_channel(rdata(31 downto 0))) :=
init_axilite_if_signals(20, 32);

  begin

      -- axilite_write overload

    procedure axilite_write(
      constant addr_value : in unsigned;
      constant data_value : in std_logic_vector;
      constant msg : in string) is
    begin
      axilite_write(addr_value, -- keep as is
                    data_value, -- keep as is
                    msg, -- keep as is
                    axi_clk, -- Clock signal
                    m_axilite_if, -- Signal must be visible in local
process scope
                    C_SCOPE, -- Just use the default
                    shared_msg_id_panel, -- Use global, shared msg_id_panel
                    C_AXILITE_BFM_CONFIG_LOCAL); -- Use locally defined
configuration or C_AXILITE_BFM_CONFIG_DEFAULT

    end;

    process

      begin

      axilite_write(unsigned(ADDR_REG), x"0000_0001", "Writting into
Address Register");  -- This procedure from UVVM checks also for the
protocol correctness of AXILITE
    end process;

--  port map  m_axilite_if signals to instantiated interface

end architectural;

qarlosalberto, 2017-08-04:
Thank you @ansariabd Im goint to try it :)

UVVM, 2017-10-17:
This issue has been fixed in the latest release.

closed 2017-10-17 by UVVM