UVVM/UVVM #26: #23 Reopen: Issue with i2c_slave_receive procedure when I2C VIP used as Master and Slave in a Test Harness

AHHarish, 2017-08-10:
Hi Marius,

I am reopening the thread again. This thread is in continuation to the old thread #23.

Please consider adding additional command as given below after the i2c_master_transmit command. Still it fails. Use the same setup what you have described.

i2c_slave_receive(I2C_VVCT, 2, 1, "One byte from master to slave");
i2c_master_transmit(I2C_VVCT, 1, C_FPGA_DEVICE_ID, x"AA", "I2C TX"); -- CHANGED by Marius
await_completion(I2C_VVCT,1,100 * 10*C_BIT_PERIOD); -- ADDED by AHHarish

Regards,
AHHarish

UVVM, 2017-08-14:
Hi,
My simulation run fine, even with the await_completion() command.
Please see my files : https://1drv.ms/f/s!AojzQFNf4hf7gcJ_1XJI-0CoQ2kAkA

Best regards,
Marius

AHHarish, 2017-08-17:
Are you using the following version. I have considered this as the latest release.
This project uses the following versions:
  uvvm_vvc_framework, version v2.0.7
  uvvm_util, version v2.1.1
  bitvis_vip_sbi, version v2.0.6
  bitvis_vip_uart, version v2.0.5
  bitvis_uart, version v1.0.4
  bitvis_irqc, version v1.0.4
  bitvis_vip_axilite, version v2.0.7
  bitvis_vip_axistream, version v3.0.5
  bitvis_vip_avalon_mm, version v2.0.7
  bitvis_vip_i2c, version v2.0.5
  bitvis_vip_spi, version v2.0.3
  bitvis_vip_gpio, version v1.0.1

Paebbels, 2017-08-17:
You can refer to a Git hash instead of publishing all single file versions :). E.g. 1fb806af5a23440dd2caf1a8fd23c22506b6e655 to point to the version from June, 8th. Or you can use tag names like [v2.4.1](https://github.com/UVVM/UVVM_All/releases/tag/v2.4.1)

This eases reproduction, because the support team can checkout the specific version you used.

UVVM, 2017-08-18:
Hi,
No, I don't get any time out issues.
Try running my setup ( https://1drv.ms/f/s!AojzQFNf4hf7ge0f0W8j5xiEpOP_WA ) by running run.sh from the /sim folder.

Br.
Marius

closed 2017-09-18 by UVVM