UVVM/UVVM #27: vvc_generator.py

roynil, 2017-08-31:
There is a typo in vvc_generator.py for READ command calling old store_result procedure:

                file_handle.write("        --     work.td_vvc_entity_support_pkg.store_result(instance_idx  => GC_INSTANCE_IDX,\n")
                file_handle.write("        --                                       cmd_idx       => v_cmd.cmd_idx,\n")
                file_handle.write("        --                                       data          => v_read_data);\n")

should be:

                file_handle.write("        --     work.td_vvc_entity_support_pkg.store_result(result_queue => result_queue,\n")
                file_handle.write("        --                                       cmd_idx       => v_cmd.cmd_idx,\n")
                file_handle.write("        --                                       result        => v_read_data);\n")

UVVM, 2017-08-31:
Hi Ronny,
Thank you for your feedback! And, you are absolutely right! We are soon to push a new release and I will ensure the error is corrected.

Best regards
Marius Elvegård, Bitvis AS

UVVM, 2017-10-17:
This issue has been fixed in the latest release.

closed 2017-10-17 by UVVM