UVVM/UVVM #29: Declaring t_axilite_if signal

stfarley, 2017-09-14:
How does one declare the the t_axilite_if signal to connect to the vvc in the test harness. The unconstrained arrays need to be constrained I believe.

I tried this.
constant GC_ADDR_WIDTH : integer  := 8;
constant GC_DATA_WIDTH : integer  := 32;
constant GC_STRB_WIDTH : integer  := GC_ADDR_WIDTH/8;

signal axi_ctrl  :t_axilite_if(
        write_address_channel.awaddr(GC_ADDR_WIDTH),  
        write_data_channel.wdata(GC_DATA_WIDTH),     
    	write_data_channel.wstrb(GC_STRB_WIDTH),     
    	read_address_channel.araddr(GC_ADDR_WIDTH),   
    	read_data_channel.ardata(GC_DATA_WIDTH));

But get 
(vcom-1356) Bad record element constraint (indexed name) for record type bitvis_vip_axilite.axilite_bfm_pkg.t_axilite_if

Is there an example AXI LITE Test Harness?

UVVM, 2017-09-15:
Hi,
No, unfortunately there is no test harness example for the AXI-Lite.
If you have a look inside the axilite_bfm_QuickRef.pdf on page two, you will see the axilite_if record type.
Then you could try to change your code to something like:
signal axi_ctrl : t_axilite_if(         write_address_channel( awaddr(GC_ADDR_WIDTH-1 downto 0)),
                                                               write_data_channel( wdata(GC_DATA_WIDTH-1 downto 0),
                                                                                                              wstrb( GC_STR_WIDTH-1 downto 0)),
                                                               read_address_channel( araddr(GC_ADDR_WIDTH-1 downto 0)),
                                                               read_data_channel( rdata(GC_DATA_WIDTH-1 downto 0)) ) := init_axilite_if_signals(GC_ADDR_WIDTH, GC_DATA_WIDTH);

Br,
Marius

closed 2017-09-15 by UVVM

stfarley, 2017-09-15:
Thanks Marius.

I had just come across that as I saw your update.
BTW I should be able to provide a simple AXILITE Test Harness example shortly.

Steve