UVVM/UVVM #30: Accessing the VVCT from a subprogram

stfarley, 2017-09-19:
Can the vvct be accessed from a procedure or function so that one could make the test sequence more hierarchical. for instance doing something like this?

procedure reg_test is
	begin
 		axilite_write(AXILITE_VVCT,1,x"0008",x"5a5a", "Writing to reg");
	end procedure reg_test ;

UVVM, 2017-09-20:
Hi,
Yes, you can access VVC commands from procedures in your TB, and making overloading procedures in the TB can be smart, e.g.

procedure axilite_write(
  constant data : in std_logic_vector(GC_DATA_WIDTH-1 downto 0);
  constant addr : in unsigned;
  constant msg  : in string := “”;
  constant VVC_INSTANCE_IDX : in natural := C_VVC_IDX_MASTER) is
begin
  axilite_write(AXILITE_VVC, VVC_INSTANCE_IDX, addr, data, msg);
end procedure

and just call: axilite_write( tx_data, tx_address );
or axilite_write( tx_data, tx_address, “Transmitting data from master to slave”);

Br.
Marius

closed 2017-09-20 by UVVM

stfarley, 2017-09-20:
Thanks Marius,

I tried putting this in a package. When I try and compile it I get either
Unknown identifier "AXILITE_VVC"
or 
Cannot drive signal "AXILITE_VVCT" from procedure

I think the latter is actually the one that needs to be used isn't it?

Here is the package.

library ieee ;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library uvvm_util;
context uvvm_util.uvvm_util_context;

library uvvm_osvvm;
use uvvm_osvvm.RandomPkg.all;
use uvvm_osvvm.CoveragePkg.all;

library uvvm_vvc_framework;
use uvvm_vvc_framework.ti_vvc_framework_support_pkg.all;

library bitvis_vip_axilite;
use bitvis_vip_axilite.vvc_methods_pkg.all;
use bitvis_vip_axilite.td_vvc_framework_common_methods_pkg.all;

package tb_pkg is
    procedure axilite_write ( 
    	constant data : in std_logic_vector(31 downto 0); 
		constant addr : in unsigned; 
		constant msg : in string := "";
		constant VVC_INSTANCE_IDX : in natural := 1) ;  
   
	procedure reg_test;  
		
end package tb_pkg;

package body tb_pkg is
	-- overload procedure so that it can be accessed within procedures
	procedure axilite_write ( 
    	constant data : in std_logic_vector(31 downto 0); 
		constant addr : in unsigned; 
		constant msg : in string := ""; 
		constant VVC_INSTANCE_IDX : in natural := 1) is  
    begin 
		axilite_write(AXILITE_VVCT, VVC_INSTANCE_IDX, addr, data, msg); 
	end procedure axilite_write; 
    
	
	procedure reg_test is

	begin
 		axilite_write(AXILITE_VVCT,1,x"0008",x"5a5a", "Writing to gpreg");
	end procedure reg_test ; 
	
	
end package body tb_pkg;

UVVM, 2017-09-21:
Hi,
You don't need the axilite_write() procedure in the pkg if you are not to use it; it was just provided as an example for smart ways to use overloads.
In the reg_test() procedure you need to add the parameter as given below, and when calling this procedure from your TB you should use reg_test(AXILITE_VVCT);

Br.
Marius

`library ieee ;
use ieee.std_logic_1164.all;
use ieee.numeric_std.all;

library uvvm_util;
context uvvm_util.uvvm_util_context;

library uvvm_vvc_framework;
use uvvm_vvc_framework.ti_vvc_framework_support_pkg.all;

library bitvis_vip_axilite;
use bitvis_vip_axilite.vvc_methods_pkg.all;
use bitvis_vip_axilite.td_vvc_framework_common_methods_pkg.all;
use bitvis_vip_axilite.td_target_support_pkg.all;

library uvvm_osvvm;
use uvvm_osvvm.RandomPkg.all;
use uvvm_osvvm.CoveragePkg.all;

package tb_pkg is

  procedure reg_test (
    signal AXILITE_VVCT : inout t_vvc_target_record
  );
  
end package tb_pkg;

package body tb_pkg is

  procedure reg_test (
    signal AXILITE_VVCT : inout t_vvc_target_record
  )is
  begin
    axilite_write(AXILITE_VVCT,1, x"0000", x"5555", "Test of axilite write");
  end procedure reg_test ;
  
end package body tb_pkg;
`

stfarley, 2017-09-21:
Thanks Marius That makes sense and is working now.