UVVM/UVVM #32: Why is the new version naming scheme not compliant to semantic versioning?

Paebbels, 2017-10-17:
May I ask why UVVM is changing its version naming scheme?

The new naming scheme is not compliant to the [semantic versioning standard](http://semver.org/). A date naming schema is violating [§6](http://semver.org/#spec-item-6), [§7](http://semver.org/#spec-item-7) and [§8](http://semver.org/#spec-item-8) of that standard.

> 6. Patch version Z (x.y.Z | x > 0) MUST be incremented if only backwards compatible bug fixes are introduced. A bug fix is defined as an internal change that fixes incorrect behavior.
> 7. Minor version Y (x.Y.z | x > 0) MUST be incremented if new, backwards compatible functionality is introduced to the public API. It MUST be incremented if any public API functionality is marked as deprecated. It MAY be incremented if substantial new functionality or improvements are introduced within the private code. It MAY include patch level changes. Patch version MUST be reset to 0 when minor version is incremented.
> 8. Major version X (X.y.z | X > 0) MUST be incremented if any backwards incompatible changes are introduced to the public API. It MAY include minor and patch level changes. Patch and minor version MUST be reset to 0 when major version is incremented.

If you fear that customers might not see when a "release" was released, please have a look at one of your older [v2.4.1 releases](https://github.com/UVVM/UVVM_All/releases/tag/v2.4.1). There is a date shown. More over you can use annotated tags to give more information about your release (like a release date).

I consider it a big drawback to jump onto the Aldec and Xilinx versioning naming scheme train just for marking reasons.

From developers stand point: The new version names are almost useless to derive the impact of UVVM changes to the own projects.

Kind regards
    Patrick

UVVM, 2017-10-17:
Hi Patrick.
Because UVVM_ALL acts as a placeholder for all the sub-repositories of UVVM (Utility library, VVC Framework and VIPs), we have decided to disconnect the version numbering of UVVM_ALL from its sub-repositories.
So, this is not done for marketing purposes, but for the reason that e.g. a major update in the SBI VIP should not have to lead to a major update of UVVM_ALL. A major update of UVVM_ALL would imply that UVVM_ALL is no longer backward compatible, but for a user that do not use the SBI VIP this update would have no impact.

Please note that all sub-repositories will continue to use the regular release numbering.

Best regards,
Marius

Paebbels, 2017-10-17:
Hello Marius,

sorry, but I don't see sub-repositories (then it would make sense). Currently I see only directories with `VERSION.txt` files. This doesn't solve the versioning problem in the sense of semantic versioning of Git repositories.

You should translate all directories into Git submodules (sub-repositories) if you want to version UVVM VIPs individually, which is by the way a very good idea.

UVVM, 2019-05-02:
Hi Patrick.
The VIPs, VVC Framework etc all follow semantic versioning, even though they are part (sub-modules) in a large repository. We think using a release date for the main repository is important to avoid a rapidly increasing version number.

closed 2019-05-02 by UVVM