UVVM/UVVM #34: How to handle your alerts?

roynil, 2017-10-27:
Hi.

I just discovered that you changed the handling of if stop-limit is reached from a assert failure to std.env.stop (line 2356): 
`assert false
               report "This single Failure line has been provoked to stop the simulation. See alert-message above"
               severity failure;`
to
`std.env.stop;`

I have used qustasims: exit -code [expr {[coverage attribute -name TESTSTATUS -concise] > 1}] to get a correct exit code for our tests but now it can't tell the difference anymore.

/Ronny

UVVM, 2017-10-27:
Hi Ronny.
Do you mean that you are not able, from within your script, to tell why the simulation stopped?
Please note that we have added the two simulation status signals:
§  - shared_uvvm_status.no_unexpected_simulation_warnings_or_worse
- shared_uvvm_status.no_unexpected_simulation_errors_or_worse
Are you able to use these in your script?
Best regards,
Marius

roynil, 2017-10-27:
Hi Marius.

1. In batch mode we rather run the design in optimized mode:
`examine sim:/methods_pkg/shared_uvvm_status`
** Error: (vish-4014) No objects found matching 'sim:/methods_pkg/shared_uvvm_status'.`

2. Even if i turn off vopt (-vopt="+acc"):
Break in Subprogram alert at /neti/asic-backup/binrepo/UVVM/uvvm_util-2.2.0/src/methods_pkg.vhd line 2356 
`examine sim:/methods_pkg/shared_uvvm_status`
{0 0}

For me a std.env.stop (or finish) represents a natural stop/finish of a simulation that's why I don't understand why you changed your alerts.

Best regards,
Ronny

UVVM, 2017-10-30:
Hi Ronny.
The reason we changed to std.env.stop in the new release is that the assert failure method was somewhat imprecise. Previously, a “failure” would be reported from the simulation, although the cause of simulation stop could be completely different, e.g. warning. We think that a stop represent a natural way of ending the simulation, even in situations when something unexpected or unwanted happened.

Br,
Marius

roynil, 2017-10-30:
Hi Marius.

But it isn't the warning itself that creates the failure. It is the fact that the number of warnings has reached its limit which for me is natural to see as a failure. 

If you intend to keep it like this I hope you could at least make this optional.

Regards,
Ronny

UVVM, 2017-11-15:
Hi Ronny,
We will soon push a new release with a switch C_USE_STD_STOP_ON_ALERT_STOP_LIMIT that you can use. Setting this to false (default is true) will end the simulation by asserting failure.

Br.
Marius

UVVM, 2017-11-16:
Switch has been added to release v2017.11.16

closed 2017-11-16 by UVVM