UVVM/UVVM #35: t_slv_array vs t_byte_array

eschmidscs, 2017-11-01:
Hi

We get ambiguity errors due to the added declaration of t_slv_array. Calls to axistream_transmit (for example) as proposed in the QuickRef with (x"00", x"00"...) are indeed not unambiguous anymore (correct me if I'm wrong here): A t_slv_array with 8-bit payload is exactly a t_byte_array.

How do you handle this collision?

Thanks and best regards,
emanuel

UVVM, 2017-11-02:
Hi Emanual,

You are absolutely right.
We have tested this with type defined arrays, which will not yield any ambiguity.
Your case with literals will as you say result in ambiguity.

We need to sort out how to fix this as there are several options (going back to only byte_array is one, moving to only slv_array is another, etc…, but they will result in different types of problems.)
In the mean time we hope you can use a workaround.
E.g. comment out  the slv-version (or byte version), use a constant or variable rather than a literal, or type qualify the literal in the procedure call.

Best regards
Marius

eschmidscs, 2017-11-03:
Hi Marius

I can definitely work with variables for now, so this is not a major issue.

In fact, I was just thinking if your solution might be that literals are just not supported. It is not a nice solution either, but it seems there is none...
And it does break backwards-compatibility. So you might as well just drop the byte_array (which mainly breaks backwards-compatibility). I'll be updating my few testbenches anyway :-)

Best regards,
emanuel

eschmidscs, 2017-11-15:
Hi Marius

As an experiment, I replaced the definition of t_byte_array with 
subtype t_byte_array  is t_slv_array(open)(7 downto 0);
This makes t_byte_array and t_slv_array the "same" type.

However, the change causes quite some more changes, because all overloaded functions are now double. Further, the shared_vvc_cmd must have a defined length, so you end up fixing the length in both the BFM and the VVC.

This works fine with questa. However, GHDL did not accept the statement above.

Best regards,
emanuel

Paebbels, 2017-11-15:
You mean the partial constraint with keyword open?

```VHDL
subtype t_byte_array is t_slv_array(open)(7 downto 0);
```

eschmidscs, 2017-11-15:
Yes, exactly.

Paebbels, 2017-11-15:
@eschmidscs 
Please see my issue report at GHDL. Your code line shows no errors.
But there is a related error as reported in the issue.

What GHDL version are you using? Do you need a latest GHDL build for Windows?

eschmidscs, 2017-11-15:
@Paebbels 
I guess I have to try this again. I used the master branch with linux.
But I didn't investigate it further... It was this late-in-the-evening-between-shower-and-bed kind of test.

I have a "stable" modified uvvm/axistream version now, so I'll put that through GHDL.

UVVM, 2017-11-15:
@eschmidscs
There is probably no way to have the byte array and slv array co-exist without ambiguity for literals.  This means we must either remove or rename one of these. Byte array has existed for a longer time, but slv is more flexible, - so we think the best solution now is to rename the byte array version to '<previous-name>_byte'.

eschmidscs, 2017-11-15:
I just realise that I didn't exactly test this. But my expectation would be that a subtype does not lead to ambiguity errors.
In fact I had to remove the t_byte_array function because having one with t_byte_array and t_slv_array lead to function redefinition errors.

So the intention was:
- Have only functions with t_slv_array
- Define t_byte_array as subtype of t_slv_array

From user-side, this should look like before. But there is no ambiguity anymore, because a) the types are assignable and b) there is only one function.

I'll give that a short try.

UVVM, 2017-11-15:
I think the question here is whether you call the procedure with a variable, signal, constant or literal (i.e. just the value like (x"A1", x"A2"...)). I think it should work perfectly well for the first three because they will be declared as either t_slv_array or t_byte_array, but not for the literal - as the compiler has no way of knowing whether (x"A1", x"A2"...) is t_slv_array or t_byte_array.
-- Espen

eschmidscs, 2017-11-15:
If t_byte_array is a subtype of t_slv_array, this doesn't matter anymore, because it is legal to assign subtypes among each other. It is perfectly ok to assign the subtype t_byte_array to a t_slv_array.

eschmidscs, 2017-11-15:
I did the test mentioned above. The problem is that std_logic_vector and t_slv_array are also ambiguous for the literal (x"00", x"01", x"02", x"03). Instead of three variants I have two now...

UVVM, 2017-11-16:
Names of all calls using t_byte_array has been changed to include suffix _bytes in release v2017.11.16

closed 2017-11-16 by UVVM