UVVM/UVVM #41: Compile script reference to sim

eschmidscs, 2017-11-20:
Hej

I'm not sure what the intention is: The compile tcl-scripts reference the sim directory for each part. This directory however is not in the repo in all cases.
I "fixed" that in some scripts by replacing "sim" with "scripts" (see scriptpath-branch in my fork).
But maybe you would prefer to have the sim-directory everywhere?

BR/emanuel

eschmidscs, 2017-11-20:
Sorry, that was pretty incomplete...

- The compile scripts for uvvm_util and uvvm_vvc_framework are ok. 
- If I use the bitvis_vip_axistream one and do an out-of-source build, i.e.:
`QuestaSim> do ~/291-11/fw/lib/uvvm/bitvis_vip_axistream/script/compile_src.do  ~/291-11/fw/lib/uvvm/bitvis_vip_axistream /tmp/foo`
the compiler complains:
`# ** Error: (vcom-7) Failed to open design unit file "/home/eschmid/291-11/fw/lib/uvvm/bitvis_vip_axistream/../bitvis_vip_axistream/sim/../src/axistream_bfm_pkg.vhd" in read mode.`

The reason seems the sim directory that exists for uvvm_util and uvvm_vvc_framework, but not for bitvis_vip_axistream. So either this directory should exist or the script should reference another path (like script).

UVVM, 2018-01-05:
Hi Emanuel,
Thank you for your feedback. This error will be fixed in  the next release.

Best regards,
Marius

closed 2018-01-05 by UVVM