UVVM/UVVM #45: axistream - Fatal Error

MMoox, 2018-02-20:
Hi everybody,

I'm using Sigasi with Riviera PRO and i ran into this error trying to simulate my design with a axistream interface:
**Fatal Error: RUNTIME_0043 axistream_bfm_pkg.vhd (1228): Value -1823774432 out of range (0 to 2147483647)**

Which refers to this code in axistream_receive() procedure in axistream_bfm_pkg.vhd:
**constant c_num_strb_bits_per_word : natural := axistream_if.tstrb'length;**

I'm using the axistream vvc in slave mode with a data width of 32 Bit, no user data, not using tdest, tkeep or tid. 
tdest, tid, tkeep are running into this or similiar errors. 

Code snippets:
[image]
_______
[image]

Please help me, i dont know how to fix this issue. 

regards,
Max

UVVM, 2018-02-21:
Hi Max.
I do not see any errors in the code and I suspect it is Riviera Pro that is having trouble with the interface being an unconstrained record.
You can try to rewrite the BFM to an non-record version and see if that will fix the error.
I will continue to investigate this fault and check with Aldec as well.

Br.
Marius

MMoox, 2018-02-21:
Hi Marius,

thank you very much for your help.
Constraining every port in the axistream bfm record solved my error.
[image]

regards,
Max

closed 2018-02-21 by MMoox