UVVM/UVVM #46: await_any_completion

roynil, 2018-02-22:
Hi,

In my sequencer I do this to wait for either channel to complete:

`await_any_completion(MY_VVCT, 0, TX, v_tx_cmd_idx, NOT_LAST, 200 us, "Await tx-transactions");
`
`await_any_completion(MY_VVCT, 0, RX, v_rx_cmd_idx,     LAST, 200 us, "Await  rx-transactions");
`
after the await_any_completion is completed I would expect:

`my_vvc_lib.vvc_methods_pkg.shared_my_vvc_status(TX, 0).previous_cmd_idx = v_tx_cmd_idx`
or 
`my_vvci_lib.vvc_methods_pkg.shared_my_vvc_status(RX, 0).previous_cmd_idx = v_rx_cmd_idx`

but neither of them has changed.

Is there any other way of telling me which channel triggered the completion of the await_any_completion?

Regards,
Ronny

UVVM, 2018-02-23:
Hi Ronny,

Could you please send the complete transcript (for this section) and the corresponding sequencer section (the commands)?
Could you also send us the actual values of *all* the variables in the example code (4 all together).

Regards 
Espen

roynil, 2018-02-23:
Sequencer and log.
[await_any.txt](https://github.com/UVVM/UVVM_All/files/1751298/await_any.txt)
[log_file.txt](https://github.com/UVVM/UVVM_All/files/1751300/log_file.txt)