UVVM/UVVM #49: function get_queue_is_full is not working properly

fergzie, 2018-05-03:
If "queue_size_in_bits" isn't divisible by "v_count" the below IF statement from ti_data_queue_pkg.vhd doesn't trigger. 

The IF statement should rather be bigger than or equal, instead of only equal.

"if v_count(queue_idx) >= v_queue_size_in_bits(queue_idx) then"

//--- Code snippet from ti_data_queue_pkg.vhd ---//
-----------------------------------------
  -- get_queue_is_full
  ------------------------------------------
  impure function get_queue_is_full(
    queue_idx : natural
    ) return boolean is
  begin
    check_value(v_queue_initialized(queue_idx), TB_WARNING, "get_queue_is_full called, but queue " & to_string(queue_idx) & " not initialized.", v_scope.all, ID_NEVER);
    if v_count(queue_idx) = v_queue_size_in_bits(queue_idx) then
      return true;
    else
      return false;
    end if;
  end function;

UVVM, 2018-05-07:
Hi,

Thank you for reporting this issue.
We will fix this in the next release.

Best regards,
Arild

UVVM, 2019-04-24:
Fix was released in v2018.07.02

closed 2019-04-24 by UVVM