UVVM/UVVM #51: axistream usage help

lukipedio, 2018-05-29:
I am trying to use an axistream BFM in my own testbench. The testbench uses axilite BFM successfully.
I have to drive a number of axistream interface (N = number of possible channels). So I declare:
    type t_axistream_array is array (0 to 63) of t_axistream_if (tdata(31 downto 0),
                                          tkeep(3 downto 0),
                                          tuser(0 downto 0),
                                          tstrb(3 downto 0),
                                          tid(7 downto 0),
                                          tdest(3 downto 0));                        
    signal axistream_if : t_axistream_array; 

Then I initualize signals: 
        for i in 0 to 63 loop
          axistream_if(i) <= init_axistream_if_signals(true,axistream_if(i).tdata'length,
                                                         axistream_if(i).tuser'length,
                                                         axistream_if(i).tid'length,
                                                         axistream_if(i).tdest'length);
          
        end loop;

Then I try to drive a master axistream interface in this way:

axistream_transmit_bytes( v_data_array ,  "Send", s_in_data_axis_aclk, axistream_if(0));
where v_data_array is 
variable v_data_array : t_byte_array(0 to 3) := (x"00", x"01", x"02", x"03");

Then I get the following message when I run the simulation:

# UVVM: ***  TB_ERROR #1  ***
# UVVM:         512.5 ns   AXISTREAM_BFM
# UVVM:                    axistream_transmit(4B) => Failed. Boolean was false. 'Sanity check: Check that bfm_config.clock_period is set'

Do I need to set clock period? Why?

UVVM, 2018-05-29:
Hi,

Yes, you need to define a config that has to be included as a parameter in the BFM procedure. Please see the documentation.
We are looking into alternatives that don't need a clock period to be set for a future release.

Best regards,
Arild

lukipedio, 2018-05-29:
Thanks Arild. What if the clock period is not set equal to the interface clock in use?