UVVM/UVVM #61: UVVM Utility Library: shared_uvvm_status variable problem

jonasppsi, 2018-08-07:
Hello,
We use the variable: `shared_uvvm_status.no_unexpected_simulation_errors_or_worse` to return an error in modelsim for script based automation purpose.

Unfortunately this variable is '0' at startup and at the end without errors/warnings it still returns '0' instead of '1'...
For a workaround I put `alert(WARNING, ...)` at the beginning. That forces an initialization of the variable to '1'.
Thanks,
Jonas

jonasppsi, 2018-08-07:
Changing the two entries in constant `C_UVVM_STATUS_DEFAULT` of `uvvm_util/src/types_pkg.vhd` to '1' would solve the problem.

UVVM, 2018-08-09:
Hi,

Thank you for reporting this issue.
After looking into this issue we have decided to make some changes in the next UVVM release.
no_unexpected_simulation_errors_or_worse and no_unexpected_simulation_warnings_or_worse will be renamed found_unexpected_simulation_errors_or_worse and found_unexpected_simulation_warnings_or_worse, hence changing the polarity.
We will also introduce new elements in the shared_uvvm_status record, mismatch_on_expected_simulation_warnings_or_worse and mismatch_on_expected_simulation_errors_or_worse, that indicate when there is a mismatch between expected and actual alerts.

Best regards,
Arild

closed 2018-09-06 by arildreiersen