UVVM/UVVM #63: IRQC demo compilation fails - bitvis sbi compile_bfm.do missing

xmentos, 2018-09-27:
I'm following the [getting started instructions](https://github.com/UVVM/UVVM/blob/master/GETTING_STARTED.md) and the IRQC demo fails because compile_bfm.do for bitvis_vip_sbi is missing. To reconstruct the problem, fetch the master branch and then from ModelSim:
`> pwd
/foo/bar/UVVM/bitvis_irqc/script
> do compile_and_sim_all.do
.. compile output ..
# ** Error: Cannot open macro file: ../../bitvis_irqc/..//bitvis_vip_sbi/script/compile_bfm.do
# Executing ONERROR command at macro ./../../bitvis_irqc/script/compile_tb_dep.do line 64
`

UVVM, 2018-09-28:
Hi,

There was a bug in the script compile_tb_dep.do.
It is fixed and should compile now.

Best regards,
Arild

xmentos, 2018-09-28:
Hi Arild, indeed it's working again now! Thanks

closed 2018-09-28 by xmentos