UVVM/UVVM #65: check_value comparing with 'Z'

RadekRR, 2018-10-18:
Hi,
I have a problem with the check_value procedure, everything works fine, when I compare '0'es and '1's but for comparing 'Z' procedure always report an error. Problem affects both std_logic and std_logic_vectors.

Here is the code snippet:
check_value('Z', 'Z', MATCH_STD, ERROR, "Compare STD 'Z' to 'Z'", msg_id     => ID_SEQUENCER);
check_value('Z', 'Z', MATCH_EXACT, ERROR, "Compare EXACT 'Z' to 'Z'", msg_id => ID_SEQUENCER);
check_value('0', '0', MATCH_STD, ERROR, "Compare STD '0' to '0'", msg_id => ID_SEQUENCER);

And results:

# UVVM: =========================================================================================================================================================================
# UVVM: ***  ERROR #1  ***
# UVVM:         27472 ns   TB seq.
# UVVM:                    check_value() => Failed. std_logic  Was 'Z'. Expected 'Z'
# UVVM:                    Compare STD 'Z' to 'Z'
# UVVM: =========================================================================================================================================================================
# UVVM: 
# UVVM: 
# UVVM: 
# UVVM: =========================================================================================================================================================================
# UVVM: ***  ERROR #2  ***
# UVVM:         27472 ns   TB seq.
# UVVM:                    check_value() => Failed. std_logic  Was 'Z'. Expected 'Z'
# UVVM:                    Compare EXACT 'Z' to 'Z'
# UVVM: =========================================================================================================================================================================
# UVVM: 
# UVVM: 
# UVVM: ID_SEQUENCER                   27472.0 ns  TB seq.                        check_value() => OK, for std_logic '0'. 'Compare STD '0' to '0''

UVVM, 2018-10-18:
Hi,

Thank you for reporting this issue.
We will look into this and hopefully release a fix within next week.

Best regards,
Arild

closed 2019-04-24 by UVVM

UVVM, 2019-04-24:
Fix was released in v2018.12.03