UVVM/UVVM #67: spi_master_transmit_and_recieve following fetch fails

stfarley, 2018-11-27:
When executing these three commands.

spi_master_transmit_and_receive(SPI_VVCT,vvc_instance_idx,spi_arg,msg);
await_completion(SPI_VVCT, vvc_instance_idx, 100 us, "Waiting for SPI Transtaction before fetching data");
fetch_result(SPI_VVCT, vvc_instance_idx, v_cmd_idx, v_data, "Fetching_data");

I get 
fetch_result(0): 'Fetching_data'. Failed. Trying to fetch result from not yet executed command or from command with no result stored

Just wondering if the fetch command is expected to work for in this instance.
Note the first two commands indicate successful completion.

UVVM, 2018-11-27:
Hi,

You are not updating the variable v_cmd_idx, so you are trying to fetch the result from the command with the index equal to whatever value this variable holds.
Insert the following line between spi_master_transmit_and_receive and await_completion:
v_cmd_idx := get_last_received_cmd_idx(SPI_VVCT, vvc_instance_idx);

Best regards,
Arild

stfarley, 2018-11-27:
Aha,

Thanks it works properly now.

Steve

stfarley, 2018-11-27:
One more question if you don’t mind.

If the transfers are 64 bit. 
i.e. 
GC_DATA_WIDTH								=> 32,
GC_DATA_ARRAY_WIDTH							=> 2,

How I retrieve the receive data from the last 32 clock cycles?

FYI,
fetch_result is not documented for spi_master_transmit_and_recieve

I guess one would assume that is the same as spi_master_recieve_only .

Thanks,
Steve

UVVM, 2018-11-28:
You just run fetch_result one more time with the same v_cmd_idx.
Thanks for notifying us about the documentation, we will fix this in the next release.

Best regards,
Arild

UVVM, 2019-04-26:
The documentation was updated in release v2018.12.03

closed 2019-04-26 by UVVM