UVVM/UVVM #69: Difference between UVVM and OSVVM

m-kru, 2018-12-10:
How does UVVM project differ from OSVVM. Is UVVM some kind of wrapper for OSVVM?

UVVM, 2018-12-10:
UVVM is a VHDL testbench infrastructure with logging, alert handling, checkers, awaiters, and several BFMs, - plus a testbench architecture for more complex DUTs, with a very structured test harness and reusable verification components, all being controlled via simple high level commands from a central test sequencer. You may use OSVVM randomisation and functional coverage seamlessly as a plugin anywhere inside UVVM. This combination is called the 'Unified VHDL Verification Methodology'
Please see https://github.com/UVVM/UVVM/blob/master/README.md for an introduction.
The recommended UVVM methodology is to use UVVM for your complete testbench, and then use OSVVM functions and procedures for randomisation and functional coverage when this is needed.

closed 2018-12-17 by UVVM

JimLewis, 2019-07-30:
OTOH, while UVVM currently uses OSVVM to do randomization and functional coverage modeling.   It is dishonest to characterize OSVVM as only doing randomization and functional coverage modeling.   OSVVM provides a complete verification frame work independent of UVVM with some overlapping capabilities - and numerous capabilities that UVVM does not have - such as memory modeling.

UVVM, 2019-07-31:
Strange interpretation of a pretty clear post. Please read the answer above more carefully. It says: "You may use OSVVM randomisation and functional coverage seamlessly as a plugin anywhere inside UVVM"  and "The recommended UVVM methodology is to use UVVM for your complete testbench, and then use OSVVM functions and procedures for randomisation and functional coverage when this is needed."
It does not say anything about OSVVM as such as the post above is information about UVVM and how OSVVM is used inside UVVM. Why on earth should we describe functionality in a totally different verification methodology...

JimLewis, 2019-07-31:
Funny.  I read the question as "What is the difference between UVVM and OSVVM".

Your answer makes it sound like OSVVM only provides randomization and functional coverage.  I mention this as others have expressed this concern to me.

UVVM, 2019-07-31:
We don't see any point in describing the functionality of OSVVM on the UVVM project, and if we tried you would most probably not have been happy with our view on OSVVM anyway, so we think it is better to answer what UVVM is and how we use OSVVM for randomisation and func. cov. This way we don't say anything bad or wrong about the OSVVM.