UVVM/UVVM #71: Extra spaces at the end of some lines.

m-kru, 2019-01-29:
I have realized that UVVM produces a lot of extra spaces at the end of lines. Are they really necessary?
2 examples:
[image]

[image]

UVVM, 2019-04-26:
Hi,

Some of the spaces are meant for easier alignment, for example in lines 16 and 17 to start the lines at the same offset, however the spaces at the end of the lines seem a bit unnecessary.

We will remove the ones in the reports, but the ones in single print outs require more work. Is this a big problem for you?

Best regards,
Erick

m-kru, 2019-05-07:
@UVVM It is not a problem for me at all. I just don't like thins that are superfluous :).

UVVM, 2019-05-08:
Hi,

I agree and we'll try to avoid this in the future. For now the extra spaces in the reports have been removed and this will be in our next release.

Best regards,
Erick

UVVM, 2019-06-07:
Hi,
This issue has been fixed in the lastest release.

Best regards,
Marius

closed 2019-06-07 by UVVM