UVVM/UVVM #72: Terminate current command behaviour

roynil, 2019-02-19:


roynil, 2019-02-19:
If i send terminate_current_command to a VVC that currently not executing a command and then next command gets terminated:
`
terminate_current_cmd(MY_VVCT, 0, "Terminate empty VVC");
write(MY_VVCT,0, 32x"0", "Write command to VVC");
read(MY_VVCT,1, "Read command to other VVC");
await_completion(MY_VVCT,1, 10 us, "This will timeout since write is terminated");

`
This is not what I would have expected.

EspenTallaksen, 2019-02-19:
Hi Ronny,

If I understand you correctly your two VVCs somehow access the same or related data.
I assume that your read() is only allowed to complete its access once it actually has data to read, and thus depends on the write() to provide that data?

We just tried a similar situation using the SBI interface (as we don’t have your VVC), and there everything worked fine.

I assume you have checked that the write() did *not* execute?
Have you tried an await_completion on the write immediately after the write()?

Could you possibly have made any changes to the await_completion() mechanism in your VVCs?

Best regards
Espen

roynil, 2019-02-19:
Hi Espen.
I realize my example was a bit unclear. My setup is: VVC(0) is a master and VVC(1) is a slave. They are connected directly to each other. So read should have been a write response from the slave to the master. 

terminate_current_cmd(MY_VVCT, 0, "Terminate empty VVC");
wait for 1 us;
write(MY_VVCT, 0, 32x"0", "Write command to VVC"); 
write_resp(MY_VVCT, 1, "Write response command to other VVC"); 
await_completion(MY_VVCT,1, 10 us, "This will timeout since write is terminated");

roynil, 2019-02-19:
This is the corresponding log:
[my_vvc.log](https://github.com/UVVM/UVVM/files/2879878/my_vvc.log)

UVVM, 2019-02-19:
Thanks. It helps a lot with a log  :-)
We think this is problem inside UVVM, and we are now testing out a few things.
Will come back to you as soon as possible.

UVVM, 2019-02-20:
Here is a preliminary fix:
Insert the following line:         uvvm_vvc_framework.ti_vvc_framework_support_pkg.reset_flag(terminate_current_cmd);
into the executor process inside VVC top level immediately after this line
work.td_vvc_entity_support_pkg.fetch_command_and_prepare_executor(v_cmd, command_queue, vvc_config, vvc_status, queue_is_increasing, executor_is_busy, C_VVC_LABELS);

And please let us know if that fixes you problem.

The permanent fix will probably be to modify the terminate_current_command procedure called from the interpreter.

roynil, 2019-02-21:
Great! It seems to work.

closed 2019-02-21 by roynil