UVVM/UVVM #73: Expected alerts may hide unexpected alerts

fergzie, 2019-03-06:
Hi,

This is just a suggestion for future improvement. 
By increasing the expected alerts limit you risk hiding unexpected errors, because there is no way of telling what kind of error was expected. Is there any way of including that kind of granularity in a future release?

UVVM, 2019-04-25:
Hi,

There is already an **alert_level** to distinguish between different types of logs. Could you give an example of what kind of granularity were you thinking about?

Kind regards,
Erick

fergzie, 2019-04-25:
To my understanding, then the Expected Alerts only indicate how many of the **alert_level** types are expected. But it does not indicate that I was expecting ERROR A & B, but that I instead got ERRORS B & C. Therefore the unexpected ERRORS B & C will be hidden to me as expected errors.

UVVM, 2019-04-26:
We don’t have a way of categorizing the errors so we would need to be able to specify the full name in order to filter them. However, you could just easily check the _Alert.txt report which contains a summary of all the alerts, even the expected ones. Also the alerts are printed in the log report.

Kind regards,
Erick

fergzie, 2019-04-29:
Hi Erick,

Ok. I'll postprocess the _Alert.txt log. 

Thank you.

Br,
David

closed 2019-04-29 by fergzie