UVVM/UVVM #74: fatal error: convert_slv_array_to_byte_array

olagrottvik, 2019-04-10:
I'm having an issue with the axistream and using t_slv_array

Calling the procedure like this to avoid ambiguity :
axistream_transmit( t_slv_array'(8x"D0", 8x"D1", 8x"D2", 8x"D3"), "test", clk, axis_if, C_SCOPE, shared_msg_id_panel, axistream_bfm_config);

Error message:
** Fatal: (vsim-3607) Slice range direction (downto) does not match slice prefix direction (to).    Time: 10104 ns  Iteration: 1  Process: /pdtp_tb/p_tb File: ../UVVM/script/../uvvm_util/script/../src/methods_pkg.vhd
Fatal error in ForLoop loop at ../UVVM/script/../uvvm_util/script/../src/methods_pkg.vhd line 4163

[image]

By replacing downto with to on line 4163 & 4164 in methods_pkg.vhd this is sorted. I can't tell if this is a user error by me or a bug?

Using questa 10.7c, uvvm_util, version v2.7.0 and bitvis_vip_axistream, version v4.0.3

olagrottvik, 2019-04-10:
So this issue is resolved when I fully constrain a t_slv_array variable with 'downto'

I guess it might be possible to add a check for T'ASCENDING to check whether to use to or downto in the function?

UVVM, 2019-04-15:
Hi,

Thank you for reporting this issue.
Yes, this can be fixed by checking slv_array(0)'ascending and either use "to" or "downto" like you said.
We will fix this in the next release.

Best regards,
Erick

UVVM, 2019-06-07:
Hi,
This issue has been fixed in the latest release.

Best regards,
Marius

closed 2019-06-07 by UVVM