UVVM/UVVM #75: axistream bfm ready configuration

fergzie, 2019-06-28:
Hi,

I have some questions and observations about **ready_low_at_word_num**, **ready_low_duration** and **ready_default_value**.

If **ready_default_value** is set to 1 then **ready_low_duration** is set to 0 and thus disabled. I would argue that the documentation gives the option to deassert the ready signal while receiving a packet using the **ready_low_at_word_num** together with the **ready_low_duration**.

If **ready_default_value** is set to 0 and **ready_low_at_word_num** is also set to 0, then the BFM waits until valid is 1 and then continues in the process. If no valid comes, then the BFM is stuck, until UVVM cmd times out. The BFM does not use the **max_wait_cycles configuration**.

But if **ready_default_value** is set to 0 and **ready_low_at_word_num** is set to > 0, then ready is set to 1 as soon as the receive command is issued. This means that the BFM times out after **max_wait_cycles configuration**.

This mixed behaviour confuses me. I would argue that the BFM should be abled to be configured in the following ways:

**ready_default_value** = 0:
If no valid arrives in the **max_wait_cycles configuration** then the BFM should time out.
If **ready_low_at_word_num** is set to 0 then it should keep ready deasserted for **ready_low_duration** when valid is asserted.
**ready_low_at_word_num** is set to > 0 then it should assert ready when valid is asserted, and then after **ready_low_at_word_num**-1 words it should deasserted ready for **ready_low_duration**.

**ready_default_value** = 1:
If no valid arrives in the **max_wait_cycles configuration** then the BFM should time out.
If **ready_low_at_word_num** is set to 0 then it should deasserted ready for **ready_low_duration** when valid is asserted.
If **ready_low_at_word_num** is set to > 0 then it should deasserted ready for 
**ready_low_duration** after **ready_low_at_word_num**-1 words when valid is asserted.

UVVM, 2019-07-02:
Hi,

That's correct, even if the **ready_default_value** is set to 1 it should be possible to deassert the ready signal by using **ready_low_at_word_num** together with the **ready_low_duration**.
The BFM should always timeout on **max_wait_cycles** if the valid signal doesn't arrive.
And agree with your interpretation of the configuration.

Thanks for the observations, we will fix this as soon as possible.

Best regards,
Erick

fergzie, 2019-07-03:
Hi Erick,

Thank you for the quick reply. I look forward to the fix.

Br,
David

UVVM, 2019-09-02:
Fixed in release v2019.09.02

closed 2019-09-02 by UVVM