UVVM/UVVM #76: Have local references in README.md

chevillotn, 2019-07-25:
Hi,
As we are using your UVVM library locally to our project, we would like to have local references in the README.md instead of a reference to the GitHub with no tag.
The code in question is for example:
```
UVVM consists currently of the following elements:
- [Utility Library](https://github.com/UVVM/UVVM/blob/master/uvvm_util/README.md)
```
it could be replaced with:
```
UVVM consists currently of the following elements:
- [Utility Library](./README.md)
```

This does not point to the release in question.
Thanks.

UVVM, 2019-07-30:
Hi,

Thanks for the observation, we will change this in our next release.

Best regards,
Erick

UVVM, 2019-09-02:
Fixed in release v2019.09.02

closed 2019-09-02 by UVVM