UVVM/UVVM #79: Copyright infringement in bitvis_vip_scoreboard/predefined_sb.vhd

JimLewis, 2019-07-31:
At FPGA Kongress, Espen revealed that the bitvis_vip_scoreboard package was created using the OSVVM ScoreboardGenericPkg documentation.   One result of your plagarism of the OSVVM interface is that the file predefined_sb.vhd contains instances of scoreboards that match those in OSVVM.   

Please be sure to attribute (aka give credit for) these to OSVVM.    It would also be correct that you indicate that generic_sb_pkg.vhd was derived/created using the OSVVM ScoreboardGenericPkg documentation and not an original idea of BitVis.

UVVM, 2019-08-01:
Hi Jim,
That was not at all what I said, and you know that. What I said was that I explicitly told our developers not to look at your code *at all*, to be absolute certain that nothing was copied even by mistake.
I am on vacation now, but will check the package when I'm back. 
In the mean time you shoud really check your own code 'AlerLogPkg.vhd'. This violates our copyright from our Utility Library in lots of places. Numerous functions and procedures are copied and modified from our Utility library, and even the name 'AlertLog' is from our system, that as the only such system used the combination of the terms 'log' and 'alert'. In other words plagarism of the first degree. 
I have noticed that you are attacking UVVM in all sorts of forums over a long period now. I do understand it is difficult to go from having a VHDL verification methodology monopoly to having a competitor that is exploding in usage, but it would be better if you concentrated on OSVVM, We could also start various discusions on Utility Library and several other areas, but I would rather spend my time working on UVVM.
I will have a closer look at the package though to check this.
- Espen

closed 2019-08-01 by UVVM

JimLewis, 2019-08-01:
Hi Espen,
Sorry to catch you on Vacation - you Norwegians seem to have a lovely amount of time off.

Looking back, I am actually surprised I did not put any attribution in the AlertLogPkg.   I did in CoveragePkg - and there was no similarity to the work from which inspired me.   I would be happy to add attribution as to the genesis of OSVVM AlertLogPkg - I had actually already been working on one.   An attribution would be appropriate in your scoreboard package and a SynthWorks copyright notice in the predefined_sb.vhd package as the text is practically verbatim WRT the corresponding OSVVM package instances. 

Honestly, before I started the AlertLogPkg, I had hoped to use the BitVis one instead, however, I needed integrated reporting from multiple models - and other sources, such as protected types - such as the ScoreBoardPkg,  So I looked around as to what there was and found the earlier works done by CWRU, Jan DeCaluwe, and Guy Eschemann to name a few.   

Did you tell ESA, who paid your for your scoreboard package, that much of the work was based on the OSVVM scoreboard documentation and that you could have simply started from that code base - as it is open source and you are allowed to make derivatives - provided of course that you leave copyright in place.

Look forward to future lively discussions.   

Best Regards,
Jim

UVVM, 2019-09-09:
Hi Jim,
Vacation is great from time to time :-)  
The UVVM scoreboard is not based on the OSVVM scoreboard at all, and to your information our generic Queue, which is really the basis for most of the UVVM Scoreboard, was implemented and released long before your Scoreboard. And as you can see from the code - nothing is is based on yours.
There was however a major misunderstanding internally regarding the user interface, where it was thought that it would be good for the users if the UVVM and OSVVM external interfaces could be as similar as possible. This idea really makes sense in many ways, but as we don't want wild allegations going far beyond the user interface, we decided to change this to a unique UVVM approach.

Looking forward to see you at FPGAworld next week :-)

Best regards
Espen