UVVM/UVVM #80: ti_generic_queue_pkg set_scope procedure broken

fergzie, 2019-08-02:
When calling set_scope with ALL_INSTANCES (which is default -2 in the adaptations_pkg.vhd) then the procedure call fails. The reason for this is line (line 556) marked with **bold** in the following code snippet from ti_generic_queue_pkg.vhd:

_if instance = ALL_INSTANCES then
      if scope'length > C_LOG_SCOPE_WIDTH then
        vr_scope := (others => scope(1 to C_LOG_SCOPE_WIDTH));
      else
        **vr_scope(instance)                    := (others =>  NUL);**
        vr_scope(instance)(1 to scope'length) := scope;
      end if;
      vr_scope_is_defined := (others => true);
    else_

The line after will also fail in the same way.

UVVM, 2019-08-05:
Hi, and thank you for your feedback!
I will have a look into this issue and add a fix for our next release.

Best regards,
Marius

UVVM, 2019-09-02:
Fixed in release v2019.09.02

closed 2019-09-02 by UVVM