UVVM/UVVM #82: Integrating UVVM with Continuous Integration - problem with simulators exit status.

m-kru, 2019-08-20:
I want to integrate UVVM with Continuous Integration. I have almost finished, but I have realized one thing about UVVM that I consider as a bug. UVVM calls `std.env.stop` when alert counter exceeds the limit. `std.env.stop` returns status code 0, and shell can't detect that simulation has failed. It should be `std.env.stop(1)`.
There are 2 places where the fix should be applied:

```
uvvm_util/src/methods_pkg.vhd
2388:                std.env.stop;

xConstrRandFuncCov/src/AlertLogPkg.vhd
676:          std.env.stop(1) ; 
702:          std.env.stop ;
```
I would really appreciate if you fix this.

UVVM, 2019-08-23:
Hi,
We will check this out.
- Espen

UVVM, 2019-08-27:
Hi, a fix will be included in the next release of UVVM.

Br.
Marius

UVVM, 2019-09-02:
Fixed in release v2019.09.02

closed 2019-09-02 by UVVM