UVVM/UVVM #84: Wrong target for first component in compile_all.do

Obliged, 2019-09-16:
I was trying to compile all uvvm components using this command:

```
MCOM_UVVM = -do "do $(PROJTOP)/../UVVM/script/compile_all.do \
					$(PROJTOP)/../UVVM/script \
					$(MSIM_SIMLIBS) \
					$(PROJTOP)/../UVVM/script/component_list.txt"
vsim -batch -do "onerror { exit -f -code 1 }" $(MCOM_UVVM) -do "exit" -logfile compile_uvvm.log
```
First I ran into an issue with compile_src.do::91-98. I was not compiling source into the default targets so there was no library at ...uvvm_util/sim/uvvm_util so vmap-ing to this location doesn't work. I feel that this piece of code can be made more generic, but I don't know what the initial issue was.

After commenting out this block I ran into a second issue. For some reason the first component in component_list.txt would not be compiled to  $(MSIM_SIMLIBS)/library_name, but rather to $PWD/library_name. A library was created at $(MSIM_SIMLIBS)/library_name, but no files were compiled to it. I'm not quite sure what's happening, but adding another vmap command after the existing one, fixes the issue.
I tested with both uvvm_util and bitvis_irqc as the first component. 

- [ ] Remove or change compile_src.do::91-98 so it works when not using default target.
- [ ] Duplicate `quietly vmap $lib_name $target_path/$lib_name` at compile_src.do::89

UVVM, 2019-09-16:
Hi,

We'll look into this as soon as possible.

Best regards,
Erick

Obliged, 2019-09-16:
Thanks, I tested using modelsim DE version 10.6c and 2019.3.

I seem to have a similar issue when performing 

```
vcom +cover=bcsef -quiet -work designlib /home/hos/pro/work/src/design/dut/Vhdl/dut.vhd
do /home/hos/pro/work/UVVM/uvvm_util/script/compile_src.do /home/hos/pro/work/UVVM/uvvm_util /data/simLibs
do /home/hos/pro/work/UVVM/uvvm_vvc_framework/script/compile_src.do /home/hos/pro/work/UVVM/uvvm_vvc_framework /data/simLibs
vcom -2008 +cover=bcsef -quiet -work supportlib /home/hos/pro/work/src/support/tb_dut/Vhdl/tb_dut.vhd
```

I'm getting this error: 
```
# ** Error: /home/hos/pro/work/src/support/tb_dut/Vhdl/tb_dut.vhd(11): (vcom-1598) Library "designlib" not found.
```

So the compiler can't find the designlib library. But if I move the calls to compile_src.do in front of designlib things start working again. Like this:

```
do /home/hos/pro/work/UVVM/uvvm_util/script/compile_src.do /home/hos/pro/work/UVVM/uvvm_util /data/simLibs
do /home/hos/pro/work/UVVM/uvvm_vvc_framework/script/compile_src.do /home/hos/pro/work/UVVM/uvvm_vvc_framework /data/simLibs
vcom +cover=bcsef -quiet -work designlib /home/hos/pro/work/src/design/dut/Vhdl/dut.vhd
vcom -2008 +cover=bcsef -quiet -work supportlib /home/hos/pro/work/src/support/tb_dut/Vhdl/tb_dut.vhd
```

UVVM, 2019-09-19:
Hi,

The first issue has been fixed and will be included in the next release. In the meanwhile you can update the lines 91-98 with the following code:
`# These two core libraries are needed by every VIP (except the IRQC and UART demos),
# therefore we should map them in case they were compiled from different directories
# which would cause the references to be in a different file.
# First check if the libraries are in the specified target path, if not, then look
# in the default UVVM structure.
if {$lib_name != "uvvm_util" && $lib_name != "bitvis_irqc" && $lib_name != "bitvis_uart"} {
  echo "Mapping uvvm_util and uvvm_vvc_framework"
  if {[file exists $target_path/uvvm_util]} {
    quietly vmap uvvm_util $target_path/uvvm_util
  } else {
    quietly vmap uvvm_util $source_path/../uvvm_util/sim/uvvm_util
  }
  if {[file exists $target_path/uvvm_vvc_framework]} {
    quietly vmap uvvm_vvc_framework $target_path/uvvm_vvc_framework
  } else {
    quietly vmap uvvm_vvc_framework $source_path/../uvvm_vvc_framework/sim/uvvm_vvc_framework
  }
}`

As for the second issue I can't seem to reproduce it. Perhaps try with this fix first and see if it still happens?

Best regards,
Erick

Obliged, 2019-09-19:
The issue still occurs on my end. 

Modelsim drops the handle to the modelsim.ini located in $MODELSIM after the first vmap command. It then copies the default modelsim.ini to $PWD and tries to continue from there. 

Editing source.do to this:
```
vlib $target_path/$lib_name
where
vmap $lib_name $target_path/$lib_name
where
vmap $lib_name $target_path/$lib_name
where
```

Provides this result:
```
# Source path: /home/obliged/pro/work/UVVM/uvvm_util
# Target path: /data/simLibs/proj
# Current directory is: /home/obliged/pro/work/proj/src/support/tb_dut/Simulation
# Project is: /home/obliged/pro/work/proj/misc/modelsim.ini
#
# Model Technology ModelSim DE vmap 2019.3 Lib Mapping Utility 2019.07 Jul 24 2019
# vmap uvvm_util /data/simLibs/proj/uvvm_util
# Modifying /home/obliged/pro/work/proj/misc/modelsim.ini
# Current directory is: /home/obliged/pro/work/proj/src/support/tb_dut/Simulation
# Project is: /opt/Mentor/2019.3/modelsim_dlx/linuxpe/../modelsim.ini
#
# Model Technology ModelSim DE vmap 2019.3 Lib Mapping Utility 2019.07 Jul 24 2019
# vmap uvvm_util /data/simLibs/proj/uvvm_util
# Copying /opt/Mentor/2019.3/modelsim_dlx/linuxpe/../modelsim.ini to modelsim.ini
# Modifying modelsim.ini
# Current directory is: /home/obliged/pro/work/proj/src/support/tb_dut/Simulation
# Project is: modelsim.ini
```
I'll let you know what I figure out with Mentor.

UVVM, 2019-09-19:
Hi,

Please do let us know, thanks!

Best regards,
Erick

Obliged, 2019-09-24:
Mentor was able to reproduce the issue. Not sure if they will fix it in future releases, but there is an acceptable workaround:

- use the `-c` option instead of `-batch` in the initial vsim call.

The 91-98 fix also worked nicely!

Thank you for the prompt help, I'll be closing this ticket.

closed 2019-09-24 by Obliged