UVVM/UVVM #85: Await_completion makes simulation run for longest timeout used

fergzie, 2019-09-16:
I have a test sequencer which does not use **std.env.stop** in the end to stop simulation. This is because I have some TCL code I need to run post simulation. Therefore I disable all clocks in the system and when all processes are done the simulation ends and the TCL continues. 

But when I use await_completion the simulation runs for the longest timeout used in any await_completion. So apparently there is some process that runs for the whole duration of the timeout when using await_completion, even though the command is finished.

UVVM, 2019-09-19:
Hi,

Are you disabling the clocks until after the simulation is completed? The only reason await_completion would timeout is because the VVC is waiting on something to happen so this could be the case if there's no clocks.
Otherwise you could try reducing the timeout for the await_completion call.

Best regards,
Erick

fergzie, 2019-09-19:
When the sequencer has reached the end I disable the clocks and then the process waits. Await_completion does **not** timeout. I'll try to give a simple example.

```
axistream_transmit()
await_completion(1 ms)
stop_clocks()
wait

```
This causes the simulation to run for 1 ms after I've disabled the clocks. If I change the timeout to 100 us, the simulation runs for 100 us after I've disabled the clocks. The await_completion does not timeout since the VVC returns with command completed.

UVVM, 2019-09-20:
Hi,

It seems the problem is with the simulator and the command `wait until <condition> for <time>`.

When you run the procedure await_completion, it uses the following timeout:
`wait until global_vvc_ack = '1' for ((v_start_time + timeout) - now);`
The procedure finishes as expected, but for some reason the waiting thread is still running so when the sequencer ends and there's no `std.env.stop`, it waits for the thread to finish.
You can verify this by using  `wait until <condition> for <time>` at the end of your sequencer.

Perhaps you could follow up with Mentor about this issue and see if this is a bug or just how it's supposed to be.

Best regards,
Erick

abyszuk, 2019-09-20:
Hi,
Just a comment from a random passer-by. 
I can also confirm I had problems with Questa when using `wait until <condition> for <time>` construct.
This problem appeared for me in a testbench which didn't use UVVM, so I think it's a simulator bug.

fergzie, 2019-09-23:
Hi Erick,

Thanks for looking in to this issue. I'll contact Mentor.

I'll get back to you when I get a reply from Mentor.

Best regards,
David

UVVM, 2019-09-24:
Hi David,

No problem and please do let us know when you get the reply.

Best regards,
Erick

fergzie, 2019-09-24:
Hi Erick,

I've gotten a reply from Mentor: 

_The reason that the simulation does not stop at <condition> is because there are other scheduled
events in the timing wheel that cannot be removed. Therefore the simulator continues to resolve
the rest of the scheduled activity - although it doesn't do much. The continuation is viewed as a
manifestation of the fact that other activity is still scheduled. It is not something that we are
planning to change_

So there is no fix planned. I'll have to find a workaround for this problem.

UVVM, 2019-09-24:
Hi David,

Thanks for the reply. Then I think this issue regarding UVVM is resolved, I'll close it.

Best regards,
Erick

closed 2019-09-24 by UVVM