UVVM/UVVM #86: Check value vs assert.

m-kru, 2019-10-10:
I wonder when you use `check_value`, and when pure VHDL `assert` when you write test benches. I feel like with `check_value` there is a lot of redundancy. For example, I have to write:
`    check_value(s_det_stable, '0', ERROR, "s_det_stable should be '0'");`
The message "s_det_stable should be '0'" is completely redundant here. 
Based on:
`    check_value(s_det_stable, '0', ERROR);`
it should be capable to print something like: `s_det_stable should be '0'`.
`check_value` also prints the message even if the condition is true.
To sum up, I am a little bit confused when to use `check_value` and when `assert`, do you have any rule of thumb?

EspenTallaksen, 2019-10-10:
Hi,

You should always use check_value when you  want to continue simulation after an alert – e.g. to continue simulation until it is finished – or until a failure or N number of errors or warnings.
VHDL assert is never recommended, but could typically be an acceptable alternative when something wrong happens and you want that to immediately terminate the simulation.
(But even then a check value is recommended)

For your example – please note that here is no way to automatically include the name of the failing signal in VHDL simulations.
Redundancy is never good.
A minimum variant of your example could be
check_value(s_det_stable, '0', ERROR, "s_det_stable");
This has no overhead as all provided info is used.

I guess a more informative variant would be
check_value(s_det_stable, '0', ERROR, "s_det_stable: should be inactive after *****");
In which case you explain what is tested or what has just happened to cause this.

You can turn off the positive acknowledge using disable_log_msg(ID_POS_ACK);   (or similar for VVCs)
The positive acknowledge is intended for initial trials or detailed overview/debugging. In normal cases the general positive acknowledge should be turned off.

Best regards
Espen

abyszuk, 2019-10-10:
> For your example – please note that here is no way to automatically include the name of the failing signal in VHDL simulations.

Huh? Can't you use one of signal attributes like `'simple_name` or `'path_name`?

m-kru, 2019-10-14:
You are right. I have checked all `*_name` predefined attributes and they return name of the procedure parameter, not actual argument. Looks like VHDL creators didn't think about this.

closed 2019-10-14 by m-kru