UVVM/UVVM #89: Unexpected gpio_set behavior

thecalicojack, 2019-11-04:
Hello,

I experienced an unexpected, and I think unintended, behavior with the gpio_set procedure from the GPIO BFM. 
If I use: `gpio_set(std_logic_vector(to_signed(2048,14)), "ADC data to +0.5", AdcDat);` everything works as expected. E.g AdcDat(11) is set to 1.
If I use: `gpio_set("00100000000000", "ADC data to +0.5", AdcDat);`; AdcDat(11) is set to 0 but AdcDat(2) is 1. So the procedure changes the bit order.
I'm pretty sure this is not at it is intended. And it took me quite a time to find the error, because I was looking for an error in the DUT.

Im using QuestaSim 10.7c

best regards
Michael

UVVM, 2019-11-05:
Hi Michael.
Thank you for your feedback. We will investigate the bug and publish a fix soon.

Best regards,
Marius

UVVM, 2019-11-22:
Fixed in latest release.

closed 2019-11-22 by UVVM