UVVM/UVVM #90: Incomplete release

thecalicojack, 2019-11-25:
Hello,

I just pulled your last release and discovered the following issues:
I'm on branch master respectively tag v2019.11.21

# Issues
## Component compile order
Compiling bitvis_vip_avalon_mm source
(vcom-1598) Library "bitvis_vip_scoreboard" not found.
## Missing files
Error: (vcom-7) Failed to open design unit file "./../bitvis_vip_uart/script/../src/monitor_cmd_pkg.vhd" in read mode.
## Missing components
bitvis_vip_gmii
bitvis_vip_hvvc_to_vvc_bridge
bitvis_vip_ethernet

best regards
Michael

UVVM, 2019-11-25:
Hi Michael!
Thank you for the information! The bug somehow slipped through the release process. A new release is on its way.

Br
Marius

UVVM, 2019-11-25:
Hi,
A new release has been published. I see you mention that you get a component compile order error for the Bitvis VIP Avalon MM. In the latest release of UVVM every VIP require Bitvis VIP Scoreboard to be compiled in addition to UVVM Util and UVVM VVC Framework.

Best regards,
Marius

thecalicojack, 2019-11-25:
Thanks for the quick fix. The compilation works fine now.
best regards
Michael

closed 2019-11-25 by thecalicojack